cortex-lab/phy #409: graphics driver crash

danieljdenman, 2015-06-25:
after opening a data set with 128 channels and 983907 spikes, i'm getting a graphics driver crash. running on a AMD Radeon HD 7450 card that came with my workstation, so maybe not an issue with a better graphics card, though if so, that's a thing to know and tell users about. 

not sure if this is related to the short-term fix for issue #355. i've changed the following parameters:
features_n_spikes_max_bg = 250000
features_n_spikes_max = 50000
features_grid_n_spikes_max = features_n_spikes_max
following y'all's suggestions.

rossant, 2015-06-25:
* do you still have a crash if you decrease those parameters?
* the split bug should be fixed in #397 so no need for high values here in principle
* can you try with another graphics card or another workstation (if possible another brand like nvidia)?
* #201 might be a solution: basically you'd have an option to enable low-quality markers that are 10x more efficient (so you can display 10x more points)

danieljdenman, 2015-06-25:
i've decreased those parameters factor 10, no crashes so far. no splitting issues with these values so far either. 

if it happens again, i can try #201, and after that try other hardware.

rossant, 2015-06-25:
actually #201 has not been done yet -- if we discover that we really need it, we'll reopen the issue and work on it...

danieljdenman, 2015-06-25:
got it. doesn't seem necessary to resolve this particular issue.

closed 2015-06-29 by rossant