cortex-lab/phy #412: snippets not working on windows under pr/397

nsteinme, 2015-06-25:
Console reports entering snippet mode when I type ":", but no commands seem to work, and now things like spacebar to change cluster don't work, even after pressing escape many times.

rossant, 2015-06-25:
do you see the status bar? the colon? do you see what you type? is it just the escape which doesn't work? does the console reports quitting snippet mode?

nsteinme, 2015-06-25:
Do not see the text appearing in the status bar. Console does not report
quitting snippet mode on escape.

rossant, 2015-06-26:
i had the same problem on windows. then I started to debug by adding a `print("hey")` here: https://github.com/kwikteam/phy/blob/master/phy/gui/dock.py#L78
then  it worked, even after i removed the print statement...
can you try it too?

nsteinme, 2015-06-26:
Hm, that didn't work for me. I also didn't see "hey" show up anywhere.

rossant, 2015-06-26:
can you try `conda update --all`? which windows version do you have exactly?

nsteinme, 2015-06-26:
Ok, I did that, nothing changed. Windows 7 Professional. What branch/pr of
phy should I be on right now? I am on 397.

rossant, 2015-06-26:
#418

nsteinme, 2015-06-26:
Ok. Still broken on that pr.

rossant, 2015-06-26:
Options:

* make a minimal reproducible example and file an issue on Qt
* (simpler) just use another set of shortcuts, with a-z, space, and a few special characters. ugly but will definitely work fine on all platforms...

nsteinme, 2015-06-26:
You're saying it's because of the colon? In that case, yeah, I vote we just
pick different shortcuts...

rossant, 2015-06-26:
No sorry, it was just a note to myself... it will be exactly the same shortcuts for you, just the underlying implementation will be different. The colon is not the problem: detecting the keys pressed after it is. Basically Qt doesn't let you detect keystrokes globally -- you have to bind an action to `a`, one to `b`, one to `c`, etc. and manually append an `a`, etc to the snippet... So I'll do that for the 26 letters, 10 numbers, + a few others (space, dash, comma...)

rossant, 2015-06-27:
OK, the PR now has a different implementation of the snippet system. I haven't tried on Windows, but chances are that it's going to work. Feel free to test @nsteinme  ...

nsteinme, 2015-06-28:
Works! Genius. Are there more options than selecting clusters by number
yet? That alone is really great, though.

rossant, 2015-06-28:
No, this is the only snippet I've implemented so far. You can add your own by adding the following to your user_settings:

```python

def change_correlogram_bins(gui, args):
    # `args` is a string with everything after the command and the space.
    # For example, typing `:cv w 100` results in `args = "w 100"`.
    # `what` is 'b' or 'w', `size` is a number
    # TODO: support setting both w and b at once.
    what, size = args.split(' ')
    size = float(size)  # in ms

    if what == 'w':
        half_width = size
        bin = None
    elif what == 'b':
        half_width = None
        bin = size
    else:
        raise ValueError("The command should be `w` or `b`.")

    for view in gui.get_views('correlograms'):
        view.change_bins(bin=bin, half_width=half_width)

cluster_manual_snippets = {
    'cv': change_correlogram_bins
}
```

I'd be happy to see what you come up with so that I can add them to the defaults!

rossant, 2015-06-28:
(please `git pull` from the PR for this example to work..)

PR #418 (2015-06-29): Improve phy command-line tool and spike detection
This PR improves the `phy` script and fixes several bugs.

* Use subparsers in `phy` CLI tool
* `phy traces` CLI script to visualize raw traces
* Show cluster group in stats panel
* Add library of probe files
* Bug fixes

## Closes

* Closes #214 
* Closes #378 
* Closes #394 
* Closes #403 
* Closes #404 
* Closes #412 
* Closes #413 
* Closes #416
* Closes #419
* Closes #421
* Closes #430
* Closes #432
* Closes #435
* Closes #439 
* Closes #441
closed 2015-06-29 by nippoo