cortex-lab/phy #414: spacebar selection wraparound

nsteinme, 2015-06-25:
Minor thing, but when using "shift-spacebar" to move backwards through the list of clusters, when it gets to the first it should wrap around to the last. This is especially relevant/useful because your newly split or merged clusters end up at the end.

rossant, 2015-06-25:
You can move to start/end of the list with `home` and `end` keys on the keyboard.

rossant, 2015-06-25:
I don't think shift+space should wrap around to the last: it may be confusing... I myself frequenty (and negligently) press space multiple times at the end (or shit+space at the beginning) to be sure I'm looking at the very last/first cluster. Having home/end keystrokes to move directly to the two ends of the list is sufficient IMO.

nsteinme, 2015-06-25:
Yes, fine - I didn't know about home/end. I think that's a better solution,
you're right.

closed 2015-06-25 by rossant