cortex-lab/phy #421: cutoff in time view

danieljdenman, 2015-06-25:
it seems only a subrange of time values are plotted in the feature views for some clusters. for example the cluster below, before and after splitting:
[image]
before splitting, there are no points after a certain time, after splitting you can see that there were in fact spikes across the whole recording.

rossant, 2015-06-25:
definitely weird! can I have a look at this dataset? (I'll also need to know what are these cluster numbers)

I'm wondering if it's some GPU bug...

danieljdenman, 2015-06-25:
sure. i saw this before pr/397 as well, just fyi. 

actually - i see it in the other dataset that i put in my drive previously:
https://drive.google.com/open?id=0B0mC-hj9DiJMfk54TE1LU1BUWUVYcDBsOFBVR2hCb05jak9COWhwUW9iU3pjOF9hLXZ3MVE

here's an example from that set. cluster 144:
[image]
and then after splitting [through a random part of the cluster]:
[image]

danieljdenman, 2015-06-25:
[also, if you'd downloaded the data before, i updated the .kwik after my manual clustering so the numbering might be different than the one you downloaded before...]

rossant, 2015-06-26:
thanks! I've found and fixed the bug in PR #418 (it was not GPU-related at all, just a stupid bug in the way spikes were subselected do match the maximum number of allowed spikes in the view)..

PR #418 (2015-06-29): Improve phy command-line tool and spike detection
This PR improves the `phy` script and fixes several bugs.

* Use subparsers in `phy` CLI tool
* `phy traces` CLI script to visualize raw traces
* Show cluster group in stats panel
* Add library of probe files
* Bug fixes

## Closes

* Closes #214 
* Closes #378 
* Closes #394 
* Closes #403 
* Closes #404 
* Closes #412 
* Closes #413 
* Closes #416
* Closes #419
* Closes #421
* Closes #430
* Closes #432
* Closes #435
* Closes #439 
* Closes #441
closed 2015-06-29 by nippoo