cortex-lab/phy #425: assertion error cluster-auto

lepmik, 2015-06-26:
Hi, when trying phy cluster-auto I get the following error

```
$phy cluster-auto test_hybrid_120sec.kwik --num-clusters-start=100
2015-06-26 08:49:28  session:359             The clustering `main` already exists -- switching to `main_0`.
Traceback (most recent call last):
  File "/home/mikkel/anaconda/envs/kwikteam/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "/home/mikkel/apps/KwikTeam/phy/phy/scripts/phy_script.py", line 226, in main
    exec_(cmd, globals(), locals())
  File "/home/mikkel/apps/KwikTeam/phy/phy/ext/six.py", line 638, in exec_
    exec("""exec _code_ in _globs_, _locs_""")
  File "<string>", line 1, in <module>
  File "<string>", line 1, in <module>
  File "/home/mikkel/apps/KwikTeam/phy/phy/scripts/phy_script.py", line 178, in run_auto
    session.cluster(clustering=clustering, **kwargs)
  File "/home/mikkel/apps/KwikTeam/phy/phy/cluster/session.py", line 360, in cluster
    kk = KlustaKwik(**kwargs)
  File "/home/mikkel/apps/KwikTeam/phy/phy/cluster/algorithms.py", line 617, in __init__
    assert 'num_starting_clusters' in kwargs
AssertionError
```

rossant, 2015-06-26:
Hi, 
can you try the PR #418 and use `cluster-auto test_hybrid_120sec.kwik 100`? (this is not necessarily the final API)

lepmik, 2015-06-26:
works :)

closed 2015-06-29 by rossant