cortex-lab/phy #426: Snippet mode doesn't work on Windows

kdharris101, 2015-06-26:
```
2015-06-26 13:24:24  base:524                Snippet mode enabled, press `escape
` to leave this mode.
Traceback (most recent call last):
  File "c:\users\kenneth\documents\github\phy\phy\gui\dock.py", line 79, in keyR
eleaseEvent
    self.emit('keystroke', e.key(), e.text())
  File "c:\users\kenneth\documents\github\phy\phy\gui\dock.py", line 86, in emit

    return self._event.emit(*args, **kwargs)
  File "c:\users\kenneth\documents\github\phy\phy\utils\event.py", line 103, in
emit
    res.append(callback(*args, **kwargs))
  File "c:\users\kenneth\documents\github\phy\phy\gui\base.py", line 517, in _on
_keystroke
    self._process_snippet(self._snippet_message)
  File "c:\users\kenneth\documents\github\phy\phy\gui\base.py", line 489, in _pr
ocess_snippet
    snippet = snippet[len(cmd):].strip()
AttributeError: 'QString' object has no attribute 'strip'
Traceback (most recent call last):
  File "c:\users\kenneth\documents\github\phy\phy\gui\dock.py", line 79, in keyR
eleaseEvent
    self.emit('keystroke', e.key(), e.text())
  File "c:\users\kenneth\documents\github\phy\phy\gui\dock.py", line 86, in emit

    return self._event.emit(*args, **kwargs)
  File "c:\users\kenneth\documents\github\phy\phy\utils\event.py", line 103, in
emit
    res.append(callback(*args, **kwargs))
  File "c:\users\kenneth\documents\github\phy\phy\gui\base.py", line 517, in _on
_keystroke
    self._process_snippet(self._snippet_message)
  File "c:\users\kenneth\documents\github\phy\phy\gui\base.py", line 489, in _pr
ocess_snippet
    snippet = snippet[len(cmd):].strip()
AttributeError: 'QString' object has no attribute 'strip'
```

Also displays \u200a\u258C

(from Max)

rossant, 2015-06-26:
duplicate of #412 
i'll debug on my windows machine

rossant, 2015-06-26:
might be more or less fixed in latest PR

rossant, 2015-06-26:
closing and keep #412 open

closed 2015-06-26 by rossant