cortex-lab/phy #432: fails poorly when raw.kwd not present

nsteinme, 2015-06-26:
I thought this was fixed before(?), but I get a million "error while loading waveform: invalid time" errors if the raw.kwd is not present, rather than being reminded about what I did wrong.

rossant, 2015-06-26:
note to self: write unit tests for this

rossant, 2015-06-26:
more generally: better handling of raw data

quoting @nippoo 

```
is there a path specified in the recordings[n]/raw/hdf5_path?
does it exist?
return(HDF5 file attribute)
is there a .dat or .bin of the same basename in the same path? (not sure how we deal with multiple recordings but k
return(.dat/bin)
is there a .raw.kwd of the same basename in the same path?
return(raw.kwd)
basically priority 1) hdf5 link 2) .dat/bin 3) .raw.kwd
```

PR #418 (2015-06-29): Improve phy command-line tool and spike detection
This PR improves the `phy` script and fixes several bugs.

* Use subparsers in `phy` CLI tool
* `phy traces` CLI script to visualize raw traces
* Show cluster group in stats panel
* Add library of probe files
* Bug fixes

## Closes

* Closes #214 
* Closes #378 
* Closes #394 
* Closes #403 
* Closes #404 
* Closes #412 
* Closes #413 
* Closes #416
* Closes #419
* Closes #421
* Closes #430
* Closes #432
* Closes #435
* Closes #439 
* Closes #441
closed 2015-06-29 by nippoo