cortex-lab/phy #433: wrong behavior when making cluster_group assignment with only one cluster selected

nsteinme, 2015-06-26:
It should move that cluster to its new place in the list, and auto-advance you to the next one in the list from where you were. E.g. In KV if you had one cluster selected and pressed "delete", it would move to MUA and you'd move to the next.

nsteinme, 2015-06-26:
Actually this is more general than I thought - when I am using the wizard with two clusters selected, I am also not getting the right behavior. When I assign best unsorted to good, nothing happens (though the console reports moving the cluster to good). 

If I merge two clusters, it shows me the new one, but pressing enter does not pin it, instead it pins the first cluster (even if I had already sent that to good).

rossant, 2015-06-26:
yeaah i changed this for merges because it was broken, but I might have forgotten to do it for cluster group changes (this is a new bug, right?)

nsteinme, 2015-06-26:
Yes, I think it worked before.

rossant, 2015-06-26:
this will be properly unit-tested after the big testing refactoring I have to do for 0.3.0

rossant, 2015-06-26:
hmm sorry this was a serious regression :( it should be all fixed in PR #418

closed 2015-06-29 by rossant