cortex-lab/phy #440: mean_probe_position incorrect

nsteinme, 2015-06-29:
Right now you are calculating the position of the cluster as:
mean ( masks * probe.positions )

But it should be:
sum ( masks * probe.positions ) / sum(masks)

[that's just the formula for center of mass]

Here's a specific line that computes it for each spike individually; you can probably use it directly if masks is the same size? I'm not sure.

masks = view.model.masks[...]
x = np.sum(masks*view.model.probe.positions[:,1],axis=1)/np.sum(masks,axis=1)

rossant, 2015-06-29:
do you feel like trying to do a pull request and become an actual phy contributor? :) Max or I can show you how do to it

nsteinme, 2015-06-29:
Ok! Will do it with max today.

PR #443 (2015-06-29): Fixing the mean_probe_position element in store
This fix returns the center of mass in the "y" dimension.

Closes #440
closed 2015-06-29 by rossant