cortex-lab/phy #445: Handle KlustaKwik2 parameters

rossant, 2015-06-29:
* How to specify them? In the `prm` file? (for 0.2.0 we have `phy spikesort` that acts as a drop-in replacement for `klusta`) 
* Should we maintain strict compatibility with the old PRM files? 
* Have KK2 parameter names and values changed compared to KK1?
* What should be the default parameter of `num_starting_clusters`?
* Should we rename `num_starting_clusters` to `n_starting_clusters` (at least in phy) for  consistency with all other parameters in phy?

nippoo, 2015-06-29:
Let's pass them in a dict in the PRM file.
No we don't need to maintain compatibility...
Yes we should rename `n_starting_clusters`,

we should have: 

```
traces = {
'data_files': ['blah.dat', 'blib.bin'],
'n_channels': 43,
'sample_rate': 30000,
'voltage_gain_multiplier': 1.0
}

spikedetekt = {
'blob': 1,
'blib' : 'foo',
'blab' = True
}

klustakwik = {
'n_starting_clusters': 42,
'blib' : 'foo',
'blab' = True
}
```

rossant, 2015-06-29:
Everyone will need to update their PRM files if we don't maintain compatibility. For example, if an old PRM file contained

```
...
sample_rate = 25000
nchannels = 129
...
```

the new one will contain:

```
spikedetekt = {
    'sample_rate': 25000,
    'n_channels': 129,
}
```

Same for KlustaKwik parameters.

Otherwise, using an old PRM file with phy will result in an error.

@nsteinme @kdharris101 what do you think? (question is: should we maintain compatibility with the PRM files between the old klustasuite and phy)

nippoo, 2015-06-29:
The new KK2 doesn't use all the same parameters... if we're going to stick with .prm for a while (which looks likely) then we might as well have nice support for different clustering algorithms etc, with the `dict`s having the same name as the module...

nsteinme, 2015-06-29:
I think breaking it is fine for me, though if it wouldn't be hard to write
a converter, why not? Maybe others would be annoyed.

rossant, 2015-06-29:
> I think breaking it is fine for me, though if it wouldn't be hard to write
> a converter, why not? Maybe others would be annoyed.

that sounds like a good idea

thesamovar, 2015-06-29:
Converter is ok but some parameters don't have equivalents so the ensemble of parameters a user specifies might not make sense in KK2. Perhaps it's better to force users to rethink parameters for the new version? I don't have strong feelings on the matter.

Default values of parameters should come from KK2, they are all specified in the parameters.py file. It should use the latest values and not be hard coded in phy I think.

We shouldn't rename ``num_starting_clusters`` in KK2 because the KK2 code consistently uses ``num_`` and not ``n_`` for quite a very variables (and generally there is a preference for slightly longer, more descriptive variable names). Happy to have a translation in phy but my personal opinion is that it feels less clean to me.

rossant, 2015-06-29:
> Default values of parameters should come from KK2, they are all specified in the parameters.py file. It should use the latest values and not be hard coded in phy I think.

agreed, maybe we can have `klustakwik = {}` in phy's user settings, meaning that the defaults come from KK2 and not phy

> We shouldn't rename num_starting_clusters in KK2 because the KK2 code consistently uses num_ and not n_ for quite a very variables (and generally there is a preference for slightly longer, more [...]

I didn't mean to rename the variable in KK2, but in phy. Contrary to all other KK2 variables, this one shows up because specifying it explicitly seems to be necessary when launching the algorithm. If there is a way to make this parameter optional (i.e. have a sensible default coded directly in KK2?), then we don't have to rename anything. The user who wants to use a specific parameter will use `num_starting_clusters` because this is a KK2-specific parameter, like all other KK2 parameters.

(I don't feel strongly about `n_` or `num_` but I chose the former because it's the standard convention in scikit-learn and other popular SciPy-based libraries)

thesamovar, 2015-06-29:
Yeah, perhaps I should change the way that works so that it's the same as all the other parameters. I'll do that: it's probably the easiest option. I'll make an issue for that.

nippoo, 2015-06-29:
Yeah, let's not write a converter. Lots of people are still using random old .prm files; let's make a really nice documented example one (or a web page) and let people upgrade. It really won't take more than about 2 minutes if people want to blindly copy values on...

`n_foo` is the defined standard in phy, rather than `num_foo`. Any other clustering/spike detection algorithms that end up within the `phy` codebase will use `n_foo` throughout so I get Cyrille's point - though I think we have bigger issues on our plate - and I think the people who notice will probably be in the single-digits! ;-)

nsteinme, 2015-06-30:
If all the parameters have different names between KK1 and KK2 - and as a
start toward building/testing the flexibility to include new algorithms -
maybe "klustakwik1 = {...}" and "klustakwik2 = {...}" should be different
parameter sets altogether, with each of those algorithms callable?

And you're all wrong - it should be "theNumberOf_foo" ;)

nippoo, 2015-06-30:
`klustakwik1` will not work within phy (it always relied on a nasty hack with writing / reading .fet/clu files etc... but yes, if it were, it would have a different parameter set!

nsteinme, 2015-06-30:
Hm but then I am confused - I thought the point was that an arbitrary
clustering algorithm could be used so long as it could accept the data and
return the results. If it needs to write those data to some location on the
hard drive, as a hack, why not let it? I'm sure there will be someone who
wants to include an algorithm that wants to do something even stranger,
don't you think?

---------- Forwarded message ----------

nippoo, 2015-06-30:
Yes, any clustering algorithm can be used so long as it:

- accepts the data as a set of NumPy arrays
- returns the data as a set of NumPy arrays
- takes a Python `dict` of parameters.

KK1 does none of the above. If someone wants to write a Python bridge program which will do the above (basically write fet/fmask, convert `dict` to a `key=value` set of arguments, execute the program, capture the logging output, then open and return the .clu), then phy makes it easy to do so, but it's not something we're planning to do!

thesamovar, 2015-06-30:
@nsteinme is there any reason to want to use KK1 over KK2? I wouldn't have thought it was worth putting any great effort into this?

nsteinme, 2015-06-30:
No, it's not worth putting in great effort. But I would say that our
comparison of the results of KK2 versus the results of KK1 has been
superficial at best so far ("yes, they both look good!" quoth I, several
times), and if we're building a system that will make it possible to do
fast/easy/automatic comparisons between algorithms or parameter sets, why
not check this? I mean, I have definitely bought all-in to KK2 already, and
am using it for everything I run. But KK1 has how many years of history of
people using it and being happy with the outcome? KK2 just doesn't have
this kind of track record yet, does it? If I were a new user of phy/kk2 and
I saw that there was a way to compare clustering algorithms - I would for
sure want to test the old thing I had been using against the new. But
again, personally I believe KK2 works great and so I fundamentally agree
that this not worth putting any great effort into it.

rossant, 2015-06-30:
@thesamovar where can I find the list of default parameter names in KK2?

thesamovar, 2015-06-30:
https://github.com/kwikteam/klustakwik2/blob/master/klustakwik2/default_parameters.py

PR #448 (2015-07-01): Bug fixes in spike detection
* New `phy spikesort` tool, replacing `klusta`
* Improved progress reporter
* Take dead channels into account during spike detection
* No more memory mapped files in store by default, but might need to check RAM consumption (?)
* Update the `PRM` format to now have `spikedetekt` and `klustakwik2` dictionary parameters

## Closes

* Closes #445 
* Closes #446
closed 2015-07-01 by nippoo