cortex-lab/phy #446: wizard not skipping MUA and Good clusters properly

nsteinme, 2015-06-29:
When I assign a cluster to MUA or Good (didn't test this with Noise), it gets the correct label but doesn't move its position in the wizard list (or get removed from the list). This means that it gets pinned (as "best unsorted") when you classify the one previous to it in the list. Probably it should pick a new best unsorted as - the next one in the list that has cluster_group=Unsorted.

rossant, 2015-06-30:
Done in the PR. Although your proposition makes sense, I'm wondering if it's not confusing? Look at the following workflow:

* A selected
* space
* B selected
* move to good
* C selected
* shift+space (previous)
* A selected

In other words, you might expect the previous cluster in the list to be the one you just moved, which it's not. The moved cluster is now at the end of the list.

nsteinme, 2015-06-30:
Yes, that is what I asked for. In what way is this different from the
behavior that Klustaviewa had? I was thinking of it in the same way.
Actually, to be exactly like KV's behavior, B would be removed from the
wizard's list altogether (the only way to select it would be to click it
directly in cluster view in KV, or use session.select in phy). But I like
the simpler "move Good/MUA/Noise to the end of the list" approach here. Was
I thinking about it the wrong way?

rossant, 2015-06-30:
no no, that's right. The slight difference is that in phy, the "best list" is *also* the entire list of clusters, including the sorted ones. Essentially because we don't have (yet) a full cluster view (well we have `:c` now)...

Just double-checking with @nippoo that this behavior is not too confusing (but I think that's fine)

nippoo, 2015-06-30:
No, I think that this behaviour makes sense. Logically you'd press Ctrl+Z if you want to undo the move, no? Then that should re-select the old one as the new best...?

nsteinme, 2015-06-30:
Yes, that sounds right.

PR #448 (2015-07-01): Bug fixes in spike detection
* New `phy spikesort` tool, replacing `klusta`
* Improved progress reporter
* Take dead channels into account during spike detection
* No more memory mapped files in store by default, but might need to check RAM consumption (?)
* Update the `PRM` format to now have `spikedetekt` and `klustakwik2` dictionary parameters

## Closes

* Closes #445 
* Closes #446
closed 2015-07-01 by nippoo