cortex-lab/phy #447: test_phy_script failure on Windows

nsteinme, 2015-06-29:
```
phy\scripts\tests\test_phy_script.py .
INTERNALERROR> Traceback (most recent call last):
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\main.py", line 84, in wrap_session
INTERNALERROR>     doit(config, session)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\main.py", line 122, in _main
INTERNALERROR>     config.hook.pytest_runtestloop(session=session)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 521, in __call__
INTERNALERROR>     return self._docall(self.methods, kwargs)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 528, in _docall
INTERNALERROR>     firstresult=self.firstresult).execute()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 394, in execute
INTERNALERROR>     res = method(*args)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\main.py", line 142, in pytest_runtestloop
INTERNALERROR>     item.config.hook.pytest_runtest_protocol(item=item, nextitem=nextitem)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 521, in __call__
INTERNALERROR>     return self._docall(self.methods, kwargs)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 528, in _docall
INTERNALERROR>     firstresult=self.firstresult).execute()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 393, in execute
INTERNALERROR>     return wrapped_call(method(*args), self.execute)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 113, in wrapped_call
INTERNALERROR>     return call_outcome.get_result()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 138, in get_result
INTERNALERROR>     py.builtin._reraise(*ex)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 123, in __init__
INTERNALERROR>     self.result = func()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 394, in execute
INTERNALERROR>     res = method(*args)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\runner.py", line 65, in pytest_runtest_protocol
INTERNALERROR>     runtestprotocol(item, nextitem=nextitem)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\runner.py", line 75, in runtestprotocol
INTERNALERROR>     reports.append(call_and_report(item, "call", log))
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\runner.py", line 121, in call_and_report
INTERNALERROR>     report = hook.pytest_runtest_makereport(item=item, call=call)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 521, in __call__
INTERNALERROR>     return self._docall(self.methods, kwargs)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 528, in _docall
INTERNALERROR>     firstresult=self.firstresult).execute()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 393, in execute
INTERNALERROR>     return wrapped_call(method(*args), self.execute)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 109, in wrapped_call
INTERNALERROR>     wrap_controller.send(call_outcome)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\skipping.py", line 157, in pytest_runtest_makereport
INTERNALERROR>     rep = outcome.get_result()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 138, in get_result
INTERNALERROR>     py.builtin._reraise(*ex)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 123, in __init__
INTERNALERROR>     self.result = func()
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\core.py", line 394, in execute
INTERNALERROR>     res = method(*args)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\runner.py", line 224, in pytest_runtest_makereport
INTERNALERROR>     longrepr = item.repr_failure(excinfo)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\python.py", line 625, in repr_failure
INTERNALERROR>     return self._repr_failure_py(excinfo, style=style)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\python.py", line 618, in _repr_failure_py
INTERNALERROR>     style=style)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\_pytest\main.py", line 410, in _repr_failure_py
INTERNALERROR>     style=style, tbfilter=tbfilter)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\py\_code\code.py", line 412, in getrepr
INTERNALERROR>     return fmt.repr_excinfo(self)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\py\_code\code.py", line 590, in repr_excinfo
INTERNALERROR>     reprtraceback = self.repr_traceback(excinfo)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\py\_code\code.py", line 582, in repr_traceback
INTERNALERROR>     reprentry = self.repr_traceback_entry(entry, einfo)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\py\_code\code.py", line 543, in repr_traceback_entry
INTERNALERROR>     s = self.get_source(source, line_index, excinfo, short=short)
INTERNALERROR>   File "C:\Anaconda\lib\site-packages\py\_code\code.py", line 484, in get_source
INTERNALERROR>     lines.append(self.flow_marker + "   " + source.lines[line_index])
INTERNALERROR> IndexError: list index out of range
```

rossant, 2015-06-29:
hmm, I didn't see this one on Windows. Can you try #448 (note that some tests are currently broken there, but the script one should pass)

also, do you have KK2 installed on your windows machine? I couldn't install it on Python 3 64-bit so the tests that rely on it all fail

rossant, 2015-07-11:
is this bug still there, or can this be closed @nsteinme @nippoo ?

nippoo, 2015-07-13:
I'm gonna close this for now.

closed 2015-07-13 by nippoo