cortex-lab/phy #449: Normalize logging display format between phy and klustakwik?

rossant, 2015-06-29:
Currently, both programs use different logging formats, which might seem weird for users:

```
2015-06-29 18:50:13  session:355             Running klustakwik...
INFO     klustakwik.initial_parameters: full_step_every = 1
```

(the first one is phy's, the second is KK's)

* Should we use the same format (at least in default INFO mode)?
* Which format to use?
    * display date+hour: yes/no?
    * display level: yes/no?

nippoo, 2015-06-29:
We should use the central phy logging system, no? Avoid duplicating code...

I would have `[DISPLAY LEVEL] [DISPLAY TIME]` (no date) so:

`18:50:30 [I] Information message here!
18:51:24 [D] Debugging message here!
`

rossant, 2015-06-29:
knowing that phy already makes an import to klustakwik2, I'd rather not have klustakwik2 depending on phy... 

We can update default logging formats in both libraries. want to do it @nippoo ?

cc @thesamovar

rossant, 2015-06-30:
@nippoo working on this now..

Another question: there are two ways to display information to the user for now: logging (INFO mode) and `print`. Should we only use one, and replace all print calls to logging INFO? Example: progress reporting when creating the cluster store or when detecting spike, `phy describe`, etc.

nippoo, 2015-06-30:
Yes, we should replace all print calls to logging INFO. This will make it much easier to redirect the output elsewhere or handle / format it differently at some later date...

rossant, 2015-06-30:
@nippoo if you look at my PR, you'll see some changes in the logging format (following your suggestion). The default format is:

```
14:40:36 [I] The automatic clustering process has finished.
```

in debug mode, the format is longer:

```
2015-06-30 14:43:48  [I]  core:394                 test 1
```

(this longer format is also used in the files)

Could you propose a PR on KK2 to match the short logging format of phy?

thesamovar, 2015-06-30:
One thing: can you set your editor not to modify whitespace? There is a wip branch with lots of changes and changing whitespace will make it harder to merge. Actually it might be better to just wait until the distributed branch is finished before changing the logging since it's not urgent.

closed 2016-03-26 by rossant