cortex-lab/phy #451: Add header in flat binary files

rossant, 2015-06-29:
Having the dtype and shape in the files will make the cluster store code much easier.

* dtype and shape as ascii string in a fixed-size header
* Remove the need to specify the dtype and shape explicitly in the store and in cluster stats

nsteinme, 2015-06-30:
You should ask @kdharris101 about this. I get the impression you mean that this would be for "internal use only" files, but if you use something more generic - specifically, like a JSON metadata file alongside the flat binary - then this achieves the same thing as what you suggest but improves readability, compatibility, and perhaps becomes the seed for a better file format down the road. It's probably worth making these decisions carefully now.

rossant, 2015-06-30:
very good point, I'm currently writing an e-mail specifically about this issue!

rossant, 2015-06-30:
e-mail sent...

rossant, 2015-07-16:
@nippoo I'm now using `.npy` for spikedetekt's cache: it's such a pain to keep track of the shape/dtype in the code. An alternative might be to use an external metadata file in JSON or YAML. But for now it's just simpler to use `.npy` (even if might be less reliable and slower).

It would be better to use the same format for the cluster store. But, this format should probably also be the same than the future kwik format, for consistency. So we'll need to settle on a single format at some point...

nippoo, 2015-07-16:
Cluster store really doesn't need to be the same as the central format: it's just a cache, and worst-case will take 20 minutes to regenerate.

I suggest we use .npy. Is it slower? I'd be very surprised. If it's slow then we should just write a loader ourselves. It's just a flat binary with a header.

rossant, 2015-07-16:
yes it's just a cache, but dealing with several *slightly* different formats in a same program is just more painful (more code, more complexity, etc.).

can you make quick benchmarks with npy (small and large arrays)? then we'll know for sure if it's a problem or not

nippoo, 2015-07-16:
npy vs flat binary?

rossant, 2015-07-16:
yes (for flat binary, you'll need to give the dtype and shape in the code)

nippoo, 2015-07-16:
just test slicing / copying the array or something?

rossant, 2015-07-16:
generate a NumPy array, save it in npy, and in flat binary. Then load both and timeit that. But you may need to take care about the OS cache...

rossant, 2015-09-27:
we'll use NPY

closed 2015-09-27 by rossant