cortex-lab/phy #454: Document how to work with various raw data formats

rossant, 2015-06-30:
* On the website (FAQ?)
* Snippet to load a raw data with a known header length (`offset` parameter)
* Snippets for common file formats

nippoo, 2015-06-30:
So - what are the common formats / acquisition systems in use for neuroscience? cc @kdharris101 @nsteinme 

So far I've got:

- flat BIN/DAT *supported*
- .raw.kwd *supported, need to check OpenEphys support #391*
- Blackrock NS5 with header (do people also use NS3, NEV, ...)?
- NeuraLynx NCS
- any more?

rossant, 2015-06-30:
maybe we could use neo (if available) for some formats: http://neo.readthedocs.org/en/latest/io.html#list-of-io

or we could just take out the code for the raw data and integrate it into phy...

nippoo, 2015-06-30:
If these systems actually define the array shape (nchannels / sample rate) and, better still, the voltage gain of the channels, it would be a shame to lose this data.

nippoo, 2015-06-30:
Ohhhh that's perfect! NEO supports a ton of different formats. Let's use that...

nsteinme, 2015-06-30:
Lots of people use/have used plexon equipment which produces .plx files or,
more recently, .pl2 files. Looks like .plx is covered by NEO but not .pl2 -
I know the person who made that format so if we want to incorporate it (or,
e.g., encourage him to write a NEO interface) at some point I can get in
touch with him.

kdharris101, 2015-06-30:
How much work would it be to incorporate this? If a lot, I would say let’s not prioritize it.

Everyone should be able to convert their data to a flat binary file – all we need is a user-editable FAQ, so that when someone with a new file format figures out how to convert it, they can write instructions…

nippoo, 2015-06-30:
I don't think this will be too much work. The question is: do we want to support reading from these formats in realtime (aka when generating waveforms etc) or should we use them exclusively to convert the file to a .dat/raw.kwd? The former avoids data duplication but might be slow depending on the format (though I don't see why it should...)

I think it would definitely be cool to be able to visualise any of these files in the TraceViewer (with live spike detection preview) before running spike detection etc.

I think import -> KWIK including spiketrains etc is probably less of a priority, except for possibly the 'legacy' .clu/.fet format (apparently this is the "KlustaKwik format" according to NEO)

closed 2016-04-13 by rossant