cortex-lab/phy #465: `phy subcommand file fails on Windows`

rossant, 2015-07-01:
doing `phy describe file.kwik` or `phy cluster-manual file.kwik` or anything with a file, fails on Windows (it shows the CLI help).
@nsteinme @nippoo can you reproduce this on Windows?

nsteinme, 2015-07-01:
No, that doesn't happen for me. I have always been using this on windows. 

[phy] C:\Users\Nick>phy describe N:/M150218_NS1LAV/20150528_all.kwik
Kwik file               N:\M150218_NS1LAV\20150528_all.kwik
Recordings              1
List of shanks          1*
Clusterings             main*, original
Channels                120
Spikes                  2240058
Clusters                636
Duration                4292s

rossant, 2015-07-03:
cannot reproduce anymore, closing for now

nippoo, 2015-07-04:
I got this error yesterday with Tim Harris. Turns out they were using system python accidentally (some environment variable was causing the script to fail etc etc). This was on stock Ubuntu 14.04.

So I think this can happen if the version of python is wrong - or maybe argparse? We should check for this...

rossant, 2015-07-04:
hmm, not sure what exactly to check, but this should definitely be investigated

rossant, 2015-07-15:
@nippoo is there a way to run a particular unit test for many versions of python? typically we could have a test for the argparse code, and test it against all versions of python 2.7.x and 3.4.x...

nippoo, 2015-07-16:
Uh - Travis?

rossant, 2015-07-16:
Can run just one test instead of the full suite?

nippoo, 2015-07-16:
We already test against 2.7 and 3.4, no?

Nothing should functionally change between minor versions, so we should assume code that runs on 2.7.0 also does on 2.7.1 etc...

rossant, 2015-07-16:
Well i think travis already tests this, so it might be a minor version pb
Or does it only happen on windows? Then we should test with appveyor?

carlilek, 2015-08-04:
I have a request from Tim Harris's lab to run this on our compute cluster, which runs Scientific Linux 6.x (redistribution of RHEL). It is our policy to install from source because of our non-standard python locations. I believe I have all of the dependencies installed, as per the Advanced Install page. 

I am seeing the error in this thread using Python 2.7.8. When I issue any command from the command line with phy (other than phy -v), I get usage back rather than any sort of error or any output. 

[carlilek@h01u32 ~]$ export PATH=/usr/local/python-2.7.8/bin:$PATH
[carlilek@h01u32 ~]$ phy -v
0.2.2.dev0
[carlilek@h01u32 ~]$

with download: 
[carlilek@h01u32 ~]$ phy --debug download hybrid_10sec.prm
usage: phy [-h] [--version] [--debug] [--hide-traceback] [--profiler]
           [--line-profiler] [--ipython] [--pdb]                     
           {download,traces,describe,spikesort,detect,cluster-auto,cluster-manual}
           ...                                                                    

Spike sorting and ephys data analysis for 1000 channels and beyond.

optional arguments:
  -h, --help            show this help message and exit
  --version, -v         print the version of phy       
  --debug, -d           activate debug logging mode (default: False)
  --hide-traceback      hide the traceback for cleaner error messages
                        (default: False)
  --profiler, -p        activate the profiler (default: False)
  --line-profiler, -lp  activate the line-profiler -- you need to decorate the
                        functions to profile with `@profile` in the code
                        (default: False)
  --ipython, -i         launch the script in an interactive IPython console
                        (default: False)
  --pdb                 activate the Python debugger (default: False)

subcommand:
  {download,traces,describe,spikesort,detect,cluster-auto,cluster-manual}
    download            download a sample dataset
    traces              show the traces of a raw data file
    describe            describe a `.kwik` file
    spikesort           launch the whole spike sorting pipeline on a `.prm`
                        file
    detect              launch the spike detection algorithm on a `.prm` file
    cluster-auto        launch the automatic clustering algorithm on a `.kwik`
                        file
    cluster-manual      launch the manual clustering GUI on a `.kwik` file

examples:
  phy -v                display the version of phy
  phy download hybrid_120sec.dat -o data/
                        download a sample raw data file in `data/`
  phy describe my_file.kwik
                        display information about a Kwik dataset
  phy spikesort my_params.prm
                        run the whole suite (spike detection and clustering)
  phy detect my_params.prm
                        run spike detection on a parameters file
  phy cluster-auto my_file.kwik
                        run klustakwik on a dataset (after spike detection)
  phy cluster-manual my_file.kwik
                        run the manual clustering GUI
[carlilek@h01u32 ~]$

rossant, 2015-08-04:
we are having a hard time reproducing this. Any way we could get remote access to this server to try to debug and fix the problem? (I'm currently in vacation but @nippoo could have a look on Thursday)

carlilek, 2015-08-04:
Unfortunately, this is not possible. If there's something you'd like me to try out and send you output from, that is no problem. 

--Ken

nippoo, 2015-08-04:
Hey, I can reproduce this on stock Ubuntu Python. Will look into it tomorrow!

carlilek, 2015-08-05:
Cool!

I actually got it working on our cluster using anaconda Python 2.7. I ran across a bug in the conda installed klustakwik, but I'll enter that over on that project.

--Ken

view it on GitHub.

nippoo, 2015-08-05:
Yeah, Anaconda Python works great. (btw - we'd recommend using Anaconda Python 3.4 if you haven't got any pressing reason to use 2.7 - and you can mix and match py2/py3 virtual environments within a single Miniconda install - though we'll be compatible with both for the foreseeable future!)

rossant, 2015-08-05:
@nippoo it would still be interesting to fix this if possible...
I think you can debug by commenting https://github.com/kwikteam/phy/blob/master/phy/scripts/phy_script.py#L34-L38

nippoo, 2015-08-05:
Yeah dw it's on the bug list for 0.2.2 - will look into it later today!

Russmilton, 2015-09-03:
Just wanted to mention that I'm getting the same issue as carlilek. I'd be happy to help you reproduce this if you still haven't done so.

closed 2015-09-09 by rossant