cortex-lab/phy #476: Create example_random_cluster.py

nippoo, 2015-07-02:
Create a file which someone can download and place in `phy/cluster/algorithms/example_random_cluster.py` or install as their own module `example_random_cluster`.

This example algorithm should take 1 parameter `n_clusters`: create that many clusters, randomly assign spikes to them, and return the clustering.

This is just to demonstrate how someone could develop their own algorithm.

rossant, 2015-07-15:
First, register your clustering function:

```python
# in an interactive session
@session.register_clustering_algorithm("myalgo")
def cluster(features=None, masks=None, spike_samples=None):
    # ...
    return spike_clusters

# in user_settings.py
def on_open(session):
    # Called when the session is opened.
    import sys
    sys.path.append('/path/to/algo/')
    import myalgo
    # myalgo.cluster(features=None, masks=None, spike_samples=None)
    # must return a vector with the cluster label for every spike (-1 for unassigned)
    session.register_clustering_algorithm("myalgo", myalgo.cluster)
```

Then, to use it:

```python
# Python API
session.cluster(..., algorithm='myalgo')

# script API
phy cluster-auto --algorithm=myalgo
```

rossant, 2015-07-15:
Further, we can have a `phy/cluster/algorithms/` with a set of built-in algorithm functions.

nippoo, 2015-07-15:
OK. This looks good.

How do we implement

1) logging (with different levels) through the algorithm
2) returning an interim value when quit early (for iterative algos like KK2 where you can just get the .temp.clu basically)
3) for bonus marks: live visualisation of clustering - related to 2)

nippoo, 2015-07-15:
4) some way of signalling whether the algorithm needs sparse features/masks or dense features/masks

rossant, 2015-07-15:
1) I'd say the logging module needs to have this capability, and then the function will use it directly. Is it sufficient to have `debug("not important stuff")` and `info("important stuff")`?
2) and 3) maybe the clustering function can `session.emit("clustering_iteration", spike_clusters=spike_clusters, some_field=some_value, ...)` regularly. Then you can do 
```
@session.connect
def on_clustering_iteration(spike_clusters=None, **kwargs):
    # visualize stuff
    # save temporary stuff
```
4) I guess for now the function can just check which type the features/masks are with `isinstance` and deal with it

nippoo, 2015-07-15:
Yeah that's ideal. (cc @shabnamkadir - fyi)

shabnamkadir, 2015-07-23:
I'm now at the point where I have preliminary superclusterings and would very much like to view the results using phy.;)

nippoo, 2015-07-23:
Simples - just update the spike_clusters array in the .kwik! :)

nippoo, 2015-07-23:
(and add a bunch more trivial clusters in the correct clustering group)

rossant, 2015-07-23:
`m = KwikModel(fn); m.add_clustering(name, spike_clusters)`

shabnamkadir, 2015-07-23:
@rossant - thanks.
@nippoo - yes, I know what needs to be done in theory...

closed 2016-03-24 by rossant