cortex-lab/phy #478: Bunch of test failures on Windows 8 64-bit

nippoo, 2015-07-02:
```
================================== FAILURES ===================================
__________________________________ test_run ___________________________________

    def test_run():
        n_spikes = 100
        with TemporaryDirectory() as tempdir:
            filename = create_mock_kwik(tempdir,
                                        n_clusters=1,
                                        n_spikes=n_spikes,
                                        n_channels=8,
                                        n_features_per_channel=3,
                                        n_samples_traces=5000)
            model = KwikModel(filename)

>           spike_clusters = run(model, num_starting_clusters=10)

build\lib\phy\cluster\tests\test_launcher.py:30:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
build\lib\phy\cluster\launcher.py:73: in run
    kk = KlustaKwik(**kwargs)
build\lib\phy\cluster\launcher.py:24: in __init__
    from klustakwik2 import __version__
..\..\klustakwik2\klustakwik2\__init__.py:4: in <module>
    from .clustering import *
..\..\klustakwik2\klustakwik2\clustering.py:12: in <module>
    from .cylib.compute_cluster_masks import accumulate_cluster_mask_sum
..\..\..\Miniconda\lib\site-packages\pyximport\pyximport.py:431: in load_module
    language_level=self.language_level)
..\..\..\Miniconda\lib\site-packages\pyximport\pyximport.py:209: in load_module
    inplace=build_inplace, language_level=language_level)
..\..\..\Miniconda\lib\site-packages\pyximport\pyximport.py:186: in build_module
    reload_support=pyxargs.reload_support)
..\..\..\Miniconda\lib\site-packages\pyximport\pyxbuild.py:100: in pyx_to_dll
    dist.run_commands()
..\..\..\Miniconda\lib\distutils\dist.py:953: in run_commands
    self.run_command(cmd)
..\..\..\Miniconda\lib\distutils\dist.py:972: in run_command
    cmd_obj.run()
..\..\..\Miniconda\lib\site-packages\Cython\Distutils\build_ext.py:163: in run
    _build_ext.build_ext.run(self)
..\..\..\Miniconda\lib\distutils\command\build_ext.py:337: in run
    self.build_extensions()
..\..\..\Miniconda\lib\site-packages\Cython\Distutils\build_ext.py:171: in build_extensions
    self.build_extension(ext)
..\..\..\Miniconda\lib\distutils\command\build_ext.py:496: in build_extension
    depends=ext.depends)
..\..\..\Miniconda\lib\distutils\msvc9compiler.py:473: in compile
    self.initialize()
..\..\..\Miniconda\lib\distutils\msvc9compiler.py:383: in initialize
    vc_env = query_vcvarsall(VERSION, plat_spec)
C:\Users\Max\Miniconda\lib\site-packages\setuptools-17.1.1-py2.7.egg\setuptools\msvc9_support.py:52: in query_vcvarsall
    return unpatched['query_vcvarsall'](version, *args, **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

version = 9.0, arch = 'amd64'

    def query_vcvarsall(version, arch="x86"):
        """Launch vcvarsall.bat and read the settings from its environment
        """
        vcvarsall = find_vcvarsall(version)
        interesting = set(("include", "lib", "libpath", "path"))
        result = {}

        if vcvarsall is None:
            raise DistutilsPlatformError("Unable to find vcvarsall.bat")
        log.debug("Calling 'vcvarsall.bat %s' (version=%s)", arch, version)
        popen = subprocess.Popen('"%s" %s & set' % (vcvarsall, arch),
                                 stdout=subprocess.PIPE,
                                 stderr=subprocess.PIPE)
[... 549 lines cut ...]

        If winsize_samples is the (odd) number of time samples in the window
        then:

            winsize_bins = 2 * ((winsize_samples // 2) // binsize) + 1
            assert winsize_bins % 2 == 1

        For performance reasons, it is recommended to compute the CCGs on a subset
        with only a few thousands or tens of thousands of spikes.

        """

        spike_clusters = _as_array(spike_clusters)
        spike_samples = _as_array(spike_samples)
        if spike_samples.dtype == np.int64:
            spike_samples = spike_samples.astype(np.uint64)

>       assert spike_samples.dtype == np.uint64
E       assert dtype('int32') == <type 'numpy.uint64'>
E        +  where dtype('int32') = array([ 0, 10, 10, 20]).dtype
E        +  and   <type 'numpy.uint64'> = <type 'numpy.uint64'>
E        +    where <type 'numpy.uint64'> = np.uint64

build\lib\phy\stats\ccg.py:86: AssertionError
```

rossant, 2015-07-02:
most of these come from KK2 being not correctly compiled no?

nippoo, 2015-07-02:
I've removed the ones that fail when KK2 isn't correctly installed now! That's another issue.

rossant, 2015-07-02:
thanks! but I'm confused because I've already fixed the remaining ones a few days ago...

nippoo, 2015-07-02:
euhhhh. I have no idea I'm afraid. Where did you fix it? The last one I'm definitely still getting when I pull master..

rossant, 2015-07-03:
all tests pass on my Windows machine, closing

closed 2015-07-03 by rossant