cortex-lab/phy #482: Fix spike detection on multiple .dat files

rossant, 2015-07-02:
```
File "/apps/phy/phy/phy/utils/array.py", line 410, in get_excerpts
if len(data) < n_excerpts * excerpt_size:
TypeError: object of type 'ConcatenatedArrays' has no len()
```

closed 2015-07-02 by nippoo

nsteinme, 2015-07-03:
Should this be reopened? It still isn't working for me on chandelier (maybe a version issue?). @nippoo let me know when I should try it.

rossant, 2015-07-03:
Is phy up to date (`git pull`)? What error message? What does `phy -v` say?

nsteinme, 2015-07-03:
Well unfortunately I'm not sure where phy is installed so I can't git pull
- is there a way to determine that? phy -v says:

0.2.0.dev0-git-v0.1.0-581-gc19a2b2b

rossant, 2015-07-03:
ok i just updated it, can you try again?

rossant, 2015-07-03:
for the record the path is `/apps/phy/phy`

nsteinme, 2015-07-03:
Great, thanks - it's running, detecting spikes. Will let you know if/when
it finishes.

PR #492 (2015-07-02): [MRG] Fix bug with multiple dat files in spike detection
Closes #482