cortex-lab/phy #500: Check for correct minimum versions of all packages on import

nippoo, 2015-07-04:
Or at least main functions.

I don't know what the best way to do this would be? But we should make sure that, at least, compatible versions of most of the libraries we use are imported.

AKA it would be much more helpful to fail with:

`ERROR: numpy 1.7 is installed, minimum required version 1.8. Please upgrade`

than 

`np has no attribute "full_like"` or whatever

nippoo, 2015-07-04:
@rossant thoughts?

This is related to #465, probably caused by an incompatible version of some library.

rossant, 2015-07-04:
it might be good to see how to do that with `pip` and `setuptools`: I'm sure there is a standard way to do it. The problem is that we **don't want** pip to install numpy and co, we want to use conda for that.

nippoo, 2015-07-04:
I mean, you could do it with a mandatory check on imports in __init__ for each module. So even if you've compiled numpy from source...!

`if np.__version__ < __NUMPY_MINIMUM_VERSION__:`

rossant, 2015-07-04:
that's a possibility, but I think it's much less clean than using a standard mechanism for dependencies in `setup.py` or whatever

nippoo, 2015-07-04:
yep if that exists it would be much nicer!

the only problem is that we have such variable dependencies - we don't need to check for, say, minimum version of `matplotlib` if we aren't planning to plot anything!

I wish there was a python syntax like `import numpy(__version__>=1.8) as np`

rossant, 2015-07-04:
can you look at the setuptools documentation`*` and see if such a thing exists?

`*` *good luck with that*

closed 2019-06-13 by rossant