cortex-lab/phy #501: Save temporary clustering of KK2 in the Kwik file

rossant, 2015-07-04:
quoting @thesamovar 

```python
def f(kk):
  clusters = kk.clusters
  ... (do what you want with the clusters)

kk = KK(...)
kk.register_callback(f)
```

This f gets called at the end of each iteration.

If you want every N iterations, inside f put:

```python
if kk.iteration%N==0
```

closed 2015-07-05 by nippoo

rossant, 2015-07-06:
@thesamovar  it seems like the callback is called many times during a split iteration? does that make sense? it looks like it's stuck in an infinite loop and the split never ends...

thesamovar, 2015-07-06:
Ah yes, my mistake I should have mentioned that. During splitting, a subset/copy KK object is created, and this inherits the callbacks. To ignore that, add the following at the beginning of the callback function:

```
if kk.name!='':
    return
```

thesamovar, 2015-07-06:
It's not stuck in an infinite loop though, it's just very slow to do the splitting.

PR #503 (2015-07-05): [MRG] Save KK2 iteration
Closes #501.

@nsteinme this saves the latest KK2 iteration in a new clustering name `kk2_current`. Due to a limitation in the file format (another one), the current active clustering cannot be changed, and it is just simpler to save it in a separate clustering.

The `KwikModel` implements several methods to manage different clusterings in the same `.kwik` file (add, remove, copy, rename, etc.).