cortex-lab/phy #515: Disparity in shape between model.masks[:] and model.features[:]

shabnamkadir, 2015-07-06:
```
model = KwikModel(kwik_path)

masky = model.masks[:] # Has shape (number of spikes, number of channels)
fetty = model.features[:]  # Has shape (number of spikes, number of features)

```

Typically number_of_channels != number of features. This is annoying for developers of, e.g. KlustaKwik.

rossant, 2015-07-06:
yeah I agree, I thought an issue was already open but I can't find it!

rossant, 2015-07-06:
actually if you want per-cluster features, you should use the cluster store (much more efficient), and that one will have the correct shape, ie `(n_spikes, n_channels, n_features)`

shabnamkadir, 2015-07-06:
At the moment I just want to convert back to the legacy format: fet and fmask files.

closed 2016-03-24 by rossant