cortex-lab/phy #531: Trouble installing phy on computing cluster

ereming, 2015-07-13:
Great to see the full phy suite is out! I've downloaded and tested it out on my Mac and it works just fine.
I would be really grateful for some guidance getting phy installed on the computing cluster I plan to use.

I can't use either the quick or advanced installation instructions because I don't have access to the root miniconda directory. So to install phy, I run the following commands:

$ conda create -n phy --clone=/cm/shared/openmind/miniconda/3.9.1
$ conda install -n phy pip numpy matplotlib scipy h5py pyqt ipython-notebook requests --yes && conda install -n phy -c http://conda.binstar.org/kwikteam klustakwik2 --yes
$ source activate phy
$ pip install vispy
$ pip install phy
$ phy download hybrid_10sec.dat
$ phy download hybrid_10sec.prm
$ phy spikesort hybrid_10sec.prm

And I get the following output:

...
All features computed and saved.
22:29:58 [I] Switched to channel group 0.
Traceback (most recent call last):
  File "/om/user/ereming/envs/phy/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/phy/scripts/phy_script.py", line 416, in main
    exec_(cmd, {}, ns)
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/phy/ext/six.py", line 638, in exec_
    exec("""exec _code_ in _globs_, _locs_""")
  File "<string>", line 1, in <module>
  File "<string>", line 1, in <module>
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/phy/cluster/session.py", line 367, in cluster
    from klustakwik2.default_parameters import default_parameters
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/__init__.py", line 3, in <module>
    from .mask_starts import *
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/mask_starts.py", line 5, in <module>
    from .numerics import mask_start_clusters
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/numerics/__init__.py", line 7, in <module>
    from .cylib import *
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/numerics/cylib/__init__.py", line 3, in <module>
    from .compute_cluster_masks import *
  File "/om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/numerics/cylib/compute_cluster_masks.py", line 1, in <module>
    from .compute_cluster_masks_cy import doaccum
ImportError: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by /om/user/ereming/envs/phy/lib/python2.7/site-packages/klustakwik2/numerics/cylib/compute_cluster_masks_cy.so)

Does this require an update to libc6 (http://askubuntu.com/questions/421642/libc-so-6-version-glibc-2-14-not-found)? If so, I'll have to request that from the cluster admins.

Along the same line, I'm wondering if this has to do with the fact that I installed using Python 2.7. I tried to install using 3, and was told that the klustakwik2 and python 3.4* conflict. python -V gives 3.4.3. I didn't have any problem installing phy with python 3 on my Mac.

rossant, 2015-07-13:
it looks like an incompatibility between the env we used to compile KK2 on linux 64-bit, and your cluster's env. In your case you could try to recompile KK2 from source directly on your cluster

```
conda uninstall klustakwik2
pip install klustakwik2
```

cc @thesamovar @nippoo

nippoo, 2015-07-13:
Sorry - the bug about the python 3.4* conflict is now solved - so you can/should try again with python 3.4! But I would recommend installing it with `pip install klustakwik2` as @rossant suggested - this is a pain on Windows, but is a reasonably foolproof way of getting it to work on Mac/Linux.

ereming, 2015-07-13:
Thanks, it's working now, after installing cython and running a pip install.

closed 2015-07-13 by ereming