cortex-lab/phy #540: phy-spikesort unexpectedly dies

nippoo, 2015-07-15:
Issue faced by Jonathan Herz.

> This happened twice to me at exactly the same spot (So I don't think I pressed anything accidentally):

> ```
> Extracting spikes: 75.61%. 11801 spikes extracted in chunk 93/122.
> Killed
> ```

> Running on 16 GB RAM

> Any idea what might be causing this error message and termination?

> Thanks
> JH

rossant, 2015-07-15:
can you check CPU and RAM usage over time, and at the time of termination?

nippoo, 2015-07-15:
@jonathanherz when you run this, could you
a) look at memory usage (run "top" in another terminal) and see if that's getting near the limit?
b) send us the contents (or last few lines) of ```/var/log/syslog``` (assuming you're on Ubuntu) to see if you're hitting the OOM killer or some other kernel issue?
c) after the script dies, run `echo $?` to see what return code it gave

My best bet is that it's running out of memory. Why, I'm not sure...

rossant, 2015-07-15:
oh and also check disk usage: spike detection should not use much RAM, but will be quite heavy on HDD/SDD

JonathanHerz, 2015-07-15:
It does look like I ran out of memory (some selected lines from /var/log/syslog):

Jul 15 01:53:24 theodore kernel: [17324.184617] Free swap  = 0kB
Jul 15 01:53:24 theodore kernel: [17324.184618] Total swap = 4104188kB

Jul 15 01:53:24 theodore kernel: [17324.184624] [ pid ]   uid  tgid total_vm      rss nr_ptes swm_sapents oocore_adj name
Jul 15 01:53:24 theodore kernel: [17324.184918] [ 3133]  1000  3133  4987138  3823327    9697   868785             0 phy

Jul 15 01:53:24 theodore kernel: [17324.184925] Out of memory: Kill process 3133 (phy) score 919 or sacrifice child
Jul 15 01:53:24 theodore kernel: [17324.184929] Killed process 3133 (phy) total-vm:19948552kB, anon-rss:15293308kB, file-rss:0kB

and echo:

jonathan@theodore:~/Documents/imec_iia_remap$ echo $?
137

I don't have a solid state drive.  Have 16GB RAM on this machine.  Are there any parameters I can change to make it less memory intensive?  The recording I'm trying to spike sort is only 2 minutes long, and I can't put any more RAM on this chipset

Thanks for looking into this, guys!

rossant, 2015-07-15:
that's weird. It looks like a memory leak, right? Does the RAM slowly increase while spike detection is running? It seems not to be the case on my computer. What is your OS exactly?

At every chunk (1 second of data) the program should release any previously allocated memory. So RAM usage should be quite low (and it is on my computer, even on large datasets), *except* when finding the thresholds initially, and potentially when computing PCA at the end of spike detection. But here it appears to happen *during* spike detection...

nippoo, 2015-07-15:
Have you got a copy of the dataset and PRB/PRM files? We could try it
here...

rossant, 2015-07-15:
you use the latest version?

JonathanHerz, 2015-07-15:
The version is 0.2.0.dev0.
What is the best way to share the files with you?  Probably too big to e-mail.
I need to go to my day job and will be back in about 6-7 hours, and can try to run it again and closely watch the memory as Cyrille suggested.

rossant, 2015-07-15:
@JonathanHerz OK, you could try again with the latest version (master), which is `0.3.0.dev0`. I did a memory fix last week that may solve your problem.

nippoo, 2015-07-15:
@JonathanHerz - the release version (0.2.0) should have this fix, so you can follow the instructions on http://phy.cortexlab.net/ to install it (you should be able to do `pip install phy`).

JonathanHerz, 2015-07-16:
still getting same behavior with both versions 0.2.1 (from pip install phy) and 0.2.2dev0 (which is what I was able to get from github).  Did not see 0.3.0.dev0.  It does look like a memory leak because I watched the System monitor each time - more and more memory was slowly gobbled up until the computer ran out of swap, then killed the process.

JonathanHerz, 2015-07-16:
using version 0.2.2.dev0 I also tried increasing my swap file size so now have 16GB RAM + 16GB swap.  After increasing swap there was enough memory for "Extracting spikes to finish".  After "Spike extraction complete" message memory was not released.  "Computing the features" also finished but once I got the "All features computed and saved" message, memory usage started going back up and OS killed phy once memory+swap was exhausted.

rossant, 2015-07-16:
Can you send us your dataset? What is your OS?

JonathanHerz, 2015-07-16:
OS is Ubuntu 14.04.  Data set is 640 MB.  What is the best way to send it over?

rossant, 2015-07-16:
Dropbox? I'd need the PRM file as well.

Do you see the same RAM issue with the sample dataset (see quick start guide)?

JonathanHerz, 2015-07-16:
Shared in dropbox - prm, dat, and prb.

Using the quick start guide it runs too fast to tell what is happening with the RAM in those steps since the data set is much smaller - I'm getting to klustakwik step in about 5 seconds.

JonathanHerz, 2015-07-16:
Ok, that's an exaggeration.  10 seconds max.

rossant, 2015-07-16:
OK thanks! Running it now on my Ubuntu 14.04 laptop. A few questions:

* on this dataset and your laptop, how long does it take for the RAM to be full?
* do you have a 64-bit system and 64-bit Python?
* do you know your versions of NumPy and Python?

rossant, 2015-07-16:
@nippoo is that possible that RAM doesn't increase on my laptop because I have a large SSD (there are a lot of memmaps under the hood)?

JonathanHerz, 2015-07-16:
Takes maybe 20-30 minutes for RAM to fill up... It is a slow and gradual process where RAM usage increases constantly.  Does not suddenly die all at once.

64 bit OS, when I fire up Python console it is 64 bit:
>>> sys.maxsize
9223372036854775807

Python version 3.4.3, Numpy version 1.9.2.  I'm using the Python that Max suggested - Anaconda or something like that?  Only using this computer for Spike Sorting, nothing else on it really.

rossant, 2015-07-16:
OK thanks. RAM usage is steady on my laptop during spike detection, and also slowly increases during spike extraction. I'm working on a fix now.

JonathanHerz, 2015-07-16:
OK thanks!  Glad you were able to reproduce the issue.

rossant, 2015-07-20:
@JonathanHerz this should be fixed by https://github.com/kwikteam/phy/pull/552 -- can you try it?

JonathanHerz, 2015-07-21:
This has been merged, right?  So if I 

git clone https://github.com/kwikteam/phy/ && cd phy && python setup.py install

that should pick up the changes, correct?  Just did that and phy -v says "0.2.2.dev0", just want to confirm I'm on the right one

Thanks

rossant, 2015-07-21:
Should be ok. @nippoo do you know why it doesnt show the full git version?

JonathanHerz, 2015-07-21:
looks like all the spike detection and extraction was done quickly and without incident.  It's now running Klustakwik

nippoo, 2015-07-21:
@JonathanHerz @rossant you have to do `python setup.py develop` to see the git version, if you do `install` then it copies the files to a new non-git-tracked location - though you're running the same code either way.

closed 2015-07-26 by rossant