cortex-lab/phy #541: Clustering fails in klusta / phy

nippoo, 2015-07-15:
Issue faced by Armin:

For background, klusta failed on this dataset. phy also seems to fail.

> Hey Max,> 

> The phy spikesort that we started yesterday is apparently finished. 2 issues :> 

> 1. the kwik file is looking very small (129 k). Is there any chance the file has been saved in another folder?> 

> 2. I again cannot open this file. Here is the error : > 
> 

> PS C:\Users\Experiment> phy cluster-manual G:\ALK01\20150604\20150604_full_phy.kwik
> 09:53:19 [I] Saving a backup of the Kwik file in G:\ALK01\20150604\20150604_full_phy.kwik.bak.
> Traceback (most recent call last):
>   File "C:\Users\Experiment\miniconda3\Scripts\phy-script.py", line 9, in <module>
>     load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 391, in main
>     out = func(args)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 337, in cluster_manual
>     use_store=not(args.no_store),
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 237, in _create_session
>     session = Session(kwik_path, **kwargs)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 85, in __init__
>     gui_classes=self._gui_classes,
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\base.py", line 382, in __init__
>     self.open(path, model=model)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 129, in open
>     return super(Session, self).open(model=model, path=kwik_path)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\base.py", line 431, in open
>     model = self._create_model(path)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 101, in _create_model
>     waveform_filter=self._waveform_filter,
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 276, in __init__
>     clustering=clustering)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 661, in open
>     self._clustering_changed(self._clustering)
>   File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 730, in _clustering_changed
>     raise ValueError("The clustering {0} is invalid.".format(value))
> ValueError: The clustering main is invalid.
> PS C:\Users\Experiment>> 
> 

> any idea? thanks
> Armin

nippoo, 2015-07-15:
Hey Armin, can you post the output of the spike sorting log - at least the last few lines?

rossant, 2015-07-15:
and also send us the kwik file?

ArminLak, 2015-07-15:
Here are last lines from the text file:

015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/29944188.waveforms.1` (float32, (192, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/29968813.waveforms.1` (float32, (306, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/29993438.waveforms.1` (float32, (220, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30018063.waveforms.1` (float32, (232, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30042688.waveforms.1` (float32, (157, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30067313.waveforms.1` (float32, (306, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30091938.waveforms.1` (float32, (351, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30116563.waveforms.1` (float32, (137, 20, 120)).
2015-07-14 05:44:18  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30141188.waveforms.1` (float32, (211, 20, 120)).
2015-07-14 05:44:20  [D]  algorithms:407           Load `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/30165813.waveforms.1` (float32, (86, 20, 120)).

also, I wonder why the kwik file is so small. It is only 129k. The dat file itself is 45G. Will email you the kwik file as I apparently cannot attach it here.

Thanks
Armin

nippoo, 2015-07-15:
Can you show more lines from the file? Ideally the whole file if it's not super-long?

ArminLak, 2015-07-15:
it is 67k lines. i am emailing it to you.

rossant, 2015-07-15:
what version did you run? there might have been a memory fix not so long ago (like last week IIRC)

rossant, 2015-07-15:
basically the process succeeded, but putting back the results into the kwik files failed. FYI, the results of the process should still sit there in your `/mnt/zserver/Data/multichanspikes/ALK01/2015-06-04/20150604_full_phy.phy/spikedetekt/` folder. There should be GBs and GBs of processed data in there :)

There is no built-in way to put this data back into the kwik files *without* reprocessing everything, but I could write one if needed.

ArminLak, 2015-07-15:
I guess Max installed the latest version 2 days ago on my computer.

I am running it again and looking at the file size. While text file is growing, kwik file stays at 129k.

rossant, 2015-07-15:
The kwik file is not expected to change during spike detection, except at the very end.

nippoo, 2015-07-15:
FYI there was a very similar error in klusta - SD completed but did not manage to put the data into the file. I think this is probably a deeper issue. No idea why

ArminLak, 2015-07-15:
yes in the .phy/spikedetekt/ there are GBs of data.  ok, I am running phy spikesort again on the data. Hopefully it will put it back into the kwik file this time.

rossant, 2015-07-20:
this should be fixed by https://github.com/kwikteam/phy/pull/552 -- can you try it?

closed 2015-07-26 by rossant