cortex-lab/phy #548: KK2 parameters not taken into account during `phy spikesort`

nippoo, 2015-07-15:
> Hey Max, did I do something wrong? I specified all sorts of things like use_mua_cluster=false and num_starting_clusters=2000 but it has used the defaults (true and 500, respectively) when it got to the clustering part of my "phy spikesort" call. Did I do something wrong or is this an unresolved issue?

> My prm is /home/nick/data/Ehrlich/20150626/params20150626_phy.prm, and I just called:
> ```(phy)nick@chandelier:~/data/Ehrlich/20150626$ phy spikesort params20150626_phy.prm```

> Thanks,

> Nick

```
nippoo@martinotti:/data/nick/Ehrlich/20150626$ cat params20150626_phy.prm 
# -*- coding: utf-8 -*-

"""Default settings for spike sorting."""

experiment_name = '20150626_all_shifted'
prb_file = '/mnt/zserver/Data/multichanspikes/Ehrlich/20150626/imecToWhisper.prb'

# -----------------------------------------------------------------------------
# Spike detection
# -----------------------------------------------------------------------------

traces = {
    'raw_data_files': ['/home/nick/data/Ehrlich/20150626/20150626_all_shifted.dat'],
    #'raw_data_files': '/mnt/zserver/Data/multichanspikes/M150212_NS1GOL/20150512/20150512_cw.dat',
    'n_channels': 129,
    'dtype': 'int16',
    'sample_rate': 25000,
}

spikedetekt = {
    #'filter_low': 500.,
    #'filter_high_factor': 0.95 * .5,  # will be multiplied by the sample rate
    #'filter_butter_order': 3,

    # Data chunks.
    #'chunk_size_seconds': 1.,
    #'chunk_overlap_seconds': .015,

    # Threshold.
    #'n_excerpts': 50,
    #'excerpt_size_seconds': 1.,
    #'use_single_threshold': True,
    'threshold_strong_std_factor': 5.5,
    'threshold_weak_std_factor': 2.,
    #'detect_spikes': 'negative',

    # Connected components.
    'connected_component_join_size': 2,

    # Spike extractions.
    'extract_s_before': 10,
    'extract_s_after': 24,
    #'weight_power': 2,

    # Features.
    #'n_features_per_channel': 3,
    #'pca_n_waveforms_max': 10000,
    
    'prb_file': '/mnt/zserver/Data/multichanspikes/Ehrlich/20150626/imecToWhisper.prb'
}

# NOTE: the default parameters are in klustakwik2's repository.
klustakwik2 = {
    'drop_last_n_features': 1,
    #'max_split_iterations': None,
    'full_step_every': 1,
    #'num_changed_threshold': 0.05,
    #'penalty_k': 0.0,
    'split_every': 50,
    #'break_fraction': 0.0,
    'always_split_bimodal': True,
    'consider_cluster_deletion': 1,
    #'subset_break_fraction': 0.01,
    #'max_quick_step_candidates': 100000000,
    'split_first': 20,
    #'mua_point': 2,
    #'max_quick_step_candidates_fraction': 0.4,
    #'dist_thresh': 9.21034037198,
    'max_iterations': 1000,
    'max_possible_clusters': 2001,
    #'penalty_k_log_n': 1.0,
    'points_for_cluster_mask': 150,
    #'fast_split': False,
    #'prior_point': 1,
    #'noise_point': 1,
    #'use_noise_cluster': True,
    'use_mua_cluster': 0,
    'num_starting_clusters': 2000,
    'save_clu_every': 10
}
```

nippoo, 2015-07-15:
cc @nsteinme

nsteinme, 2015-07-15:
Thanks - in the meantime, is there a workaround? It appears that there is nothing generated for this dataset (20150626_all_shifted) in the _klustakwik directory - I assume that step has been phased out - but in that case I'm not sure how to call KK2. Can I call it directly on a kwik file with command line parameters? The KK2 site seems to still only list the kk2_legacy method of calling it? Thanks! cc @thesamovar

nippoo, 2015-07-15:
does `phy cluster-auto myfile.prm` work?

nsteinme, 2015-07-15:
No, I get this:

(phy)nick@chandelier:~/data/Ehrlich/20150626$ phy cluster-auto params20150626_shifted.prm 
Traceback (most recent call last):
  File "/apps/miniconda/envs/phy/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
  File "/apps/phy/phy/phy/scripts/phy_script.py", line 391, in main
    out = func(args)
  File "/apps/phy/phy/phy/scripts/phy_script.py", line 300, in cluster_auto
    assert args.file.endswith('.kwik')
AssertionError

Presumably it is only taking a kwik file as input. The name of the prm file ("params20150626_shifted.prm") is not the same name as the kwik file that this params file produced ("20150626_all_shifted.kwik"). I guess there are two ways this behavior could work: it could look into the prm file, determine the kwik file name that should exist if this same prm file was used to create it, and then proceed with that one. Or cluster-auto could accept both a kwik filename and a prm file, in which case it just takes the clustering parameters from the prm file and ignores the rest. I guess the current intended behavior is a third thing - that the parameters are supposed to have been written into the kwik file so that you can call cluster-auto on the kwik file and it will use the parameters? But that is less flexible anyway, so one of these two methods involving re-specification of the parameters via a prm file should be possible.

rossant, 2015-07-15:
oops this is a bug, try deleting that line 300

PR #552 (2015-07-20): [MRG] Fix spikedetekt cache
* Better cache directory structure
* Use `.npy` format in the cache
* Save connected components in the cache
* Refactored `SpikeDetekt` class
* New `SpikeDetektStore` class, which handles saving/loading of temporary data during spike detection
* Waveforms are no longer saved in temporary files
* Delete filtered/components files once they're no longer required
* Fixed flake8 bug
* Fixed KK2 params bug
* Added integration test for quick start guide
* Various bug fixes

## Closes

* Closes #546
* Closes #547
* Closes #548
closed 2015-07-20 by nippoo