cortex-lab/phy #553: data not saved on kwik file at the end of phy spikesort

ArminLak, 2015-07-17:
Hi,

I have now tried 3 times to use phy spikesort. At the end of sorting, the kwik file is very small and it looks like sorted spikes have not been stored there. Anyway we can fix this issue? 

here is the messages in my terminal :  

Computing the features: 99.99%. chunk 7399/7400.
All features computed and saved.
Segmentation fault (core dumped)
(phy)armin@martinotti:~/data/ALK01/2015-06-04$

here is the last lines the text file: 

2015-07-17 04:45:00  [D]  algorithms:407           Load `/data/armin/ALK01/2015-06-04/20150604_full_phyv2.phy/spikedetekt/89044188.waveforms.1` (float32, (4710, 20, 120)).
2015-07-17 04:45:01  [D]  algorithms:407           Load `/data/armin/ALK01/2015-06-04/20150604_full_phyv2.phy/spikedetekt/89068813.waveforms.1` (float32, (5337, 20, 120)).
2015-07-17 04:45:02  [D]  algorithms:407           Load `/data/armin/ALK01/2015-06-04/20150604_full_phyv2.phy/spikedetekt/89093438.waveforms.1` (float32, (4905, 20, 120)).

rossant, 2015-07-17:
is this the same problem as in #541 ?

nippoo, 2015-07-17:
Ah there's a segfault there! What's that about?

rossant, 2015-07-17:
curious! very likely an HDF5 problem. what is the version of h5py?

nippoo, 2015-07-17:
2.5.0, on 64-bit Linux

rossant, 2015-07-17:
try 2.4?

nippoo, 2015-07-17:
OK I've downgraded it on the `phy` venv - @ArminLak can you try again?

If this doesn't work I have a sneaky suspicion that it's something to do with zserver and permissions / corruption or something, so try running it straight off your homedirectory on `basket`...

rossant, 2015-07-17:
also you can try the `--interval=0,10` option to only run on a subset...

rossant, 2015-07-20:
this should be fixed by https://github.com/kwikteam/phy/pull/552 -- can you try it?

ArminLak, 2015-07-20:
I am now sorting using downgraded version of h5phy (2.4). will see if it works.

ArminLak, 2015-07-20:
... and I am running it straight off my homedirectory on basket.

ArminLak, 2015-07-21:
I ran it with downgraded version of h5phy on my homedirectory and again the same error. The kwik file is 129k and I fail  to open it with phy cluster-manual. 

you mentioned this can be fixed by #552 . I dont really understand what I am supposed to do. Any hint?

Thank you

rossant, 2015-07-21:
I'm not sure how phy was installed for you... @nippoo?

nippoo, 2015-07-21:
@ArminLak you can try again (sorry!) with the version on the servers (`source activate phy`). If it doesn't work then we should try running it ourselves with your data...

Max

ArminLak, 2015-07-21:
running it again now.

ArminLak, 2015-07-22:
sorting is done. On the positive side, this time the kwik file is large. On the negative side, I still fail to open thing using phy cluster-manual. Here is the error I am getting:

PS G:\ALK01\20150604> phy cluster-manual 20150604_full_phyv2.kwik
Traceback (most recent call last):
  File "C:\Users\Experiment\miniconda3\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 391, in main
    out = func(args)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 337, in cluster_manual
    use_store=not(args.no_store),
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 237, in _create_sessio
    session = Session(kwik_path, **kwargs)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 85, in __init__
    gui_classes=self._gui_classes,
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\base.py", line 382, in __init__
    self.open(path, model=model)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 129, in open
    return super(Session, self).open(model=model, path=kwik_path)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\base.py", line 431, in open
    model = self._create_model(path)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\cluster\session.py", line 101, in _create_model
    waveform_filter=self._waveform_filter,
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 276, in __init__
    clustering=clustering)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 661, in open
    self._clustering_changed(self._clustering)
  File "C:\Users\Experiment\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 730, in _clustering_changed
    raise ValueError("The clustering {0} is invalid.".format(value))
ValueError: The clustering main is invalid.
PS G:\ALK01\20150604>

rossant, 2015-07-22:
can you put the dataset on zserver and give me the URL?

ArminLak, 2015-07-22:
I put .dat , .prm, .prb and kwik and kwx files here:

\\zserver\Data\multichanspikes\ALK01\ForCyrille

rossant, 2015-07-22:
thanks! downloading now, I'm going to have a look

rossant, 2015-07-22:
I spent two hours trying to save the file but unfortunately it appears to be corrupted. Are you sure the spike sorting process finished successfully?

ArminLak, 2015-07-22:
I see. I think it was finished successfully. I assume you have been trying on the kwik file. right ?

rossant, 2015-07-22:
Yes, I have. @nippoo do you want to have a look too? I may have missed something...

Also, it seems that this particular dataset has been problematic in the past. Can you try to use the `--interval=0,60` option (for example) to run the algorithm on a subset of the file only (in the example, the first minute)?

ArminLak, 2015-07-22:
I am doing now with interval = 0, 60 . will let you know.

nsteinme, 2015-07-23:
Hey, I'm getting the same problem as Armin - a recent phy spikesort run won't load in phy cluster-manual, giving a "The clustering main is invalid." I don't believe there was any error earlier in the process.

rossant, 2015-07-23:
does `phy describe file.kwik` work?

rossant, 2015-07-23:
otherwise, try this in IPython:

```
import h5py
f = h5py.File('file.kwik')
```

does this raise an exception?

nsteinme, 2015-07-23:
phy describe returns the same error. However the ipython lines you gave me seem to run just fine; I get an HDF5 file object.

nsteinme, 2015-07-23:
By the way, in my case at least the KK2 part is still running on these files on the server, and is still reporting that it updates the kk2_current clustering in the file. Maybe the main clustering never gets created properly if you don't complete clustering successfully after doing spike detection? Or maybe after you start kk2 it messes up the main clustering? Is there a way to start cluster-manual on a different clustering so I can see if the kk2_current one is not broken?

rossant, 2015-07-23:
the file is still running? then you cannot open it...

nsteinme, 2015-07-23:
I copied the kwik, dat, and kwx to my local hard drive, so kk2 is not
running on my local copy. Is it still true that I should not be able to
open it? If so is that something fixable? Currently this is the only way to
check on the progress of something that's maybe taking a week or more to
run, so far as I understand.

rossant, 2015-07-23:
that's the problem with saving every iteration: if you copy while the current iteration is being saved to kk2_current, then the file might be corrupted. If it's not corrupted, you need to create the main clustering out of the kk2_current clustering.

Finally, you cannot open the HDF5 file while KK2 is running.

rossant, 2015-07-23:
This is a major problem with HDF5, unfortunately. We're currently trying to find an alternative format...

nsteinme, 2015-07-23:
So are you suggesting that I just happened to copy it during the brief moment that kk2 was writing a new clustering into the file? Or are you saying that the file is held open for the entire time that kk2 is running? I thought one thing we had resolved from klusta was to always close files again as fast as possible. Related question: There was some discussion of how to make kk2 do a "graceful halt" - did that get implemented? How do I do it? It sounds like if I do ctrl+c then I will end up corrupting my kwik file?

rossant, 2015-07-23:
> So are you suggesting that I just happened to copy it during the brief moment that kk2 was writing a new clustering into the file? Or are you saying that the file is held open for the entire time [...]

Yes, files are closed asap. So it could happen that you copied the file during the brief moment that kk2 was writing a new clustering into the file. But the fact that h5py opens the file might suggest that it's not the case (?). How long is each iteration? When I do tests on simple datasets, the copy in the file appears to take a very significant portion of the processing time.

> There was some discussion of how to make kk2 do a "graceful halt" - did that get implemented? How do I do it? It sounds like if I do ctrl+c then I will end up corrupting my kwik file?

Currently, killing the process and hoping a clustering is not being written at that moment! Nothing else has been implemented in KK2 afaik.

The file format problems with HDF5 are getting extremely urgent, so we need to solve them before we can fix the kk2_clustering issues...

nsteinme, 2015-07-23:
I ran a "--interval=0,60" test and it still tells me that the main clustering is invalid. KK2 is still running (again) and I will let it finish and try again. But maybe there was a bug introduced recently that is causing the clustering to be invalid?

/home/nick/data/Ehrlich/20150626/20150626_all_shifted_segment.kwik is the kwik file associated with this interval run (26MB).

rossant, 2015-07-23:
if you `phy describe` what do you see in the list of clusterings? do you see the `main` one? it should be created at the very end of KK2.

nsteinme, 2015-07-23:
Aha, I see, so there doesn't even exist any "main" clustering before KK2
finishes?? This is a serious problem - kk2 often takes forever and you
should be able to load the file in phy before it finishes. Why not create a
"main" clustering full of 1's at the end of spikedetekt, then replace it
with the kk2 results if/when they come?

And to answer your question, phy describe also still doesn't work:

[phy] N:\Ehrlich>phy describe 20150626_all_shifted_segment.kwik
Traceback (most recent call last):
  File "C:\Anaconda\envs\phy\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line
384,
 in main
    out = func(args)
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line
241,
 in describe
    model = KwikModel(path, clustering=args.clustering)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\model.py", line 290,
in _
_init__
    clustering=clustering)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\model.py", line 662,
in o
pen
    self._clustering_changed(self._clustering)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\model.py", line 731,
in _
clustering_changed
    raise ValueError("The clustering {0} is invalid.".format(value))
ValueError: The clustering main is invalid.

rossant, 2015-07-23:
yeah, that's what I originally tried, but I got some file corruption problems :( HDF5 does not support deletion well
I'll try a slightly different way...

nsteinme, 2015-07-23:
Here's another idea then - could phy describe and phy cluster-manual look for any valid clustering before crashing? Because after just a few minutes there should be a valid "kk2_current" clustering, and if it just goes straight to this when it's the only valid one there then it would be fine. 

But I'm a little confused - aren't you overwriting the "main" clustering all the time during cluster-manual? I don't see why it would be a problem to put it there at the end of spikedetekt full of 1's and then overwrite later after kk2 finishes - the number of elements should be the same.

rossant, 2015-07-23:
> Here's another idea then - could phy describe and phy cluster-manual look for any valid clustering before crashing? Because after just a few minutes there should be a valid "kk2_current" clustering, and if it just goes straight to this when it's the only valid one there then it would be fine.

yes, that's also what I tried but I got corruption problems (I guess I was just working with a corrupted file at that time!!) -- I'll try again. Can you send me your data, or tell me where I can find it on the servers? (the fact that h5py opens it means that's it's likely not corrupted -- although that's not 100% sure)

> But I'm a little confused - aren't you overwriting the "main" clustering all the time during cluster-manual? I don't see why it would be a problem to put it there at the end of spikedetekt full of 1's and then overwrite later after kk2 finishes - the number of elements should be the same.

that would work if we only saved the `spike_clusters` array; unfortunately, we *also* have one HDF5 group per cluster, which in the case of SD2 would be overwritten (the `1` cluster).
in cluster-manual we never delete anything, your old clusters are always there for backup
that's one of the many things we need to fix in the file format

nsteinme, 2015-07-23:
/home/nick/data/Ehrlich/20150626/20150626_all_shifted_segment.kwik is the kwik file, though you will have to wait a few minutes since I just re-started it to attempt to use a kk2 subset run (so that kk2 would complete much faster, hopefully). I will let you know when it's ready, maybe 10 minutes. 

But in the way that cluster-manual will leave an "abandoned" cluster or two when merging existing ones, can't you just "abandon" clusterID==1 (i.e. leave it with 0 spikes) when adding the new main after kk2? Or just call it clusterID=-99 or something?

nsteinme, 2015-07-23:
Ok, that file is ready.

rossant, 2015-07-23:
there is no clustering at all in that file, is that normal?

```
Kwik file               /data/nick/Ehrlich/20150626/20150626_all_shifted_segment.kwik
Recordings              1
List of shanks          1*
Clusterings             empty*
Channels                120
Spikes                  237379
Clusters                1
Duration                2760s
```

nsteinme, 2015-07-23:
??? You just told me that the clustering wouldn't be created until after kk2 finished! 

Also how did you get phy describe to work on it? That crashes for me with "clustering main is invalid".

nsteinme, 2015-07-23:
p.s. looks like "duration" does not report correctly when using the "interval" option, that file was generated with --interval=0,60.

rossant, 2015-07-23:
> ??? You just told me that the clustering wouldn't be created until after kk2 finished!

* Has KK2 finished or not? Sorry I haven't understood
* There is an empty clustering by default (the code doesn't handle the case when there is no clustering at all -- this should be fixed)
* At every iteration, the kk2_current clustering is saved. So it should always be there if at least one iteration has run
* I used `KwikModel(fn).describe()` in IPython

nsteinme, 2015-07-23:
KK2 is not finished. The whole point here, for me, is to try to see what the data looks like without waiting a week... There also wasn't a single kk2 iteration run there because I was trying (but failed) to run kk2 with subset enabled (issue on this in a moment). 

I don't know if it is normal for there to be no clustering; I haven't seen this problem before. I just ran phy spikesort with the params20150626_phy.prm that you can find in that same folder.

rossant, 2015-07-23:
Ok, well look I think saving a new clustering in the kwik file at every iteration brings more problems than it solves (because of the file format). Let's switch to something different:

* At every iteration, `yourexperiment.phy/klustakwik/iterXX.txt` is saved. This is a text file with one integer (cluster id) per line. Files are never going to be corrupted, or the worse thing that can happen is that the last one is corrupted: just use the previous one.
* You can integrate the clustering into the kwik file at any time. You can already to this manually in IPython with `kwik_model.add_clustering(name, spike_clusters)`. I can give you the complete code, and you can save it in a script `add_clustering.py`.
* When you run cluster-manual, just specify `--clustering=mytempclustering`.
* I also need to fix the `main` clustering absence problem

How does that sound?

nsteinme, 2015-07-23:
Sounds great! This saving of the .txt and having a function to add it
sounds, actually, identical to the old system of the temp.clu (which was
fine). But what if we add one thing that would save a lot of headache for
me? What if there is a keyboard shortcut that interrupts kk2 and adds that
last clustering to the kwik file automatically - is that easy to do? And a
minor point but since kk2 can run for thousands of iterations with millions
of spikes, saving a .txt every iteration is not always going to be
desirable - what about just the two most recent (i.e. the one you want plus
a backup)?

Good to know about the cluster-manual --clustering=name option!

Nick

rossant, 2015-07-23:
* we can just save the last 2 or 3 iterations
* not sure about the interrupt shortcut. But you can have something pretty close: `ctrl+c` to kill the process, then `python add_clustering.py` which automatically adds the last iteration to the kwik file and launch the GUI on that clustering

nsteinme, 2015-07-23:
Ok! I suppose I can manage ;)

nsteinme, 2015-07-24:
Just to confirm here - after kk2 did finish, I was able to open the file in phy without error. @ArminLak, when you saw the original error you reported, were you sure that kk2 had finished running successfully?

ArminLak, 2015-07-24:
Sorry for my delay. I was away in a meeting.  So I ran phy spiikesort on interval=0,60 and I can use cluster manual to open it.

So I think kk2 was not finished running successfully when I was doing the whole data but I dont know what has happened. I will try again and see

rossant, 2015-07-26:
closing for now, feel free to reopen if needed

closed 2015-07-26 by rossant