cortex-lab/phy #561: klustakwik2 settings not all making it properly from the prm file to the execution of kk2

nsteinme, 2015-07-23:
It seems that use_mua_cluster doesn't get set correctly, subset doesn't get set at all, whereas points_for_cluster_mask and num_starting_clusters do get set correctly.

My .prm file has this: 
```
klustakwik2 = {
    'drop_last_n_features': 1,
    #'max_split_iterations': None,
    'full_step_every': 1,
    #'num_changed_threshold': 0.05,
    #'penalty_k': 0.0,
    'split_every': 50,
    #'break_fraction': 0.0,
    'always_split_bimodal': True,
    'consider_cluster_deletion': 1,
	'subset': 50,
    #'subset_break_fraction': 0.01,
    #'max_quick_step_candidates': 100000000,
    'split_first': 20,
    #'mua_point': 2,
    #'max_quick_step_candidates_fraction': 0.4,
    #'dist_thresh': 9.21034037198,
    'max_iterations': 1000,
    'max_possible_clusters': 2001,
    #'penalty_k_log_n': 1.0,
    'points_for_cluster_mask': 150,
    #'fast_split': False,
    #'prior_point': 1,
    #'noise_point': 1,
    #'use_noise_cluster': True,
    'use_mua_cluster': False,
    'num_starting_clusters': 2000,
    'save_clu_every': 10
}
```
My .log file shows this:
```
2015-07-23 18:02:24  [D]  logging:110      Running SpikeDetekt with the following parameters: {'consider_cluster_deletion': 1, 'use_single_threshold': 1, 'connected_component_join_size': 2, 'probe_adjacency_list': {[omitted]}, 'sample_rate': 25000, 'points_for_cluster_mask': 150, 'subset': 50, 'always_split_bimodal': 1, 'n_excerpts': 50, 'num_starting_clusters': 2000, 'n_features_per_channel': 3, 'chunk_size_seconds': 1.0, 'split_every': 50, 'raw_data_files': '/home/nick/data/Ehrlich/20150626/20150626_all_shifted.dat', 'waveform_dc_offset': '', 'threshold_weak_std_factor': 2.0, 'full_step_every': 1, 'extract_s_before': 10, 'filter_high_factor': 0.47499999999999998, 'split_first': 20, 'chunk_overlap_seconds': 0.014999999999999999, 'filter_butter_order': 3, 'extract_s_after': 24, 'dtype': 'int16', 'use_mua_cluster': 0, 'waveform_scale_factor': '', 'detect_spikes': 'negative', 'filter_low': 500.0, 'probe_channels': [etc]
```

But my kk2 output shows this:
```
18:22:46 [I] Running klustakwik...
INFO     klustakwik.initial_parameters: consider_cluster_deletion = 1
INFO     klustakwik.initial_parameters: split_first = 20
INFO     klustakwik.initial_parameters: mua_point = 2
INFO     klustakwik.initial_parameters: noise_point = 1
INFO     klustakwik.initial_parameters: max_quick_step_candidates = 100000000
INFO     klustakwik.initial_parameters: max_quick_step_candidates_fraction = 0.4
INFO     klustakwik.initial_parameters: always_split_bimodal = 1
INFO     klustakwik.initial_parameters: num_starting_clusters = 2000
INFO     klustakwik.initial_parameters: dist_thresh = 9.21034037198
INFO     klustakwik.initial_parameters: penalty_k = 0.0
INFO     klustakwik.initial_parameters: full_step_every = 1
INFO     klustakwik.initial_parameters: use_noise_cluster = True
INFO     klustakwik.initial_parameters: max_split_iterations = None
INFO     klustakwik.initial_parameters: fast_split = False
INFO     klustakwik.initial_parameters: use_mua_cluster = True
INFO     klustakwik.initial_parameters: break_fraction = 0.0
INFO     klustakwik.initial_parameters: prior_point = 1
INFO     klustakwik.initial_parameters: points_for_cluster_mask = 150
INFO     klustakwik.initial_parameters: penalty_k_log_n = 1.0
INFO     klustakwik.initial_parameters: split_every = 50
INFO     klustakwik.initial_parameters: subset_break_fraction = 0.01
INFO     klustakwik.initial_parameters: max_iterations = 1000
INFO     klustakwik.initial_parameters: max_possible_clusters = 2001
INFO     klustakwik.initial_parameters: num_changed_threshold = 0.05
INFO     klustakwik.initial_parameters: use_noise_cluster = True
INFO     klustakwik.initial_parameters: use_mua_cluster = 0
```

Note that there are evidently two "use_mua_cluster" parameters that kk2 is trying to use. cc @thesamovar .

rossant, 2015-07-27:
so weird that the parameter is present twice in the initial parameters! @thesamovar any ideas?

thesamovar, 2015-07-31:
OK I think this should be fixed now.

thesamovar, 2015-07-31:
You want me to do another release of KK2?

thesamovar, 2015-07-31:
OK, I did a 0.2.3 release with this fixed in it.

rossant, 2015-07-31:
thanks @thesamovar !
@nsteinme let us know if that fixed it, if so you can close this issue

nsteinme, 2015-08-10:
Sorry for taking a while to test this. I believe it does not seem to be fixed. First - it's a little difficult to answer this because there is no log file produced of the KK2 output anymore, is that true? If there is I'm not sure where to look for it. So I just have to watch the terminal carefully... 

But I think it's not working because my parameters file has 'subset': 50, and 'full_step_every': 3, but it's doing neither of those things. "subset" does not show up in the list of klustakwik.initial_parameters at all (and indeed it is trying to use all 2.5million spikes), and full_step_every = 1. num_starting_clusters =2000 despite that my prm file specifies 500, however points_for_cluster_mask=150, correctly, which I think is not the default value. 

Is it possible that it is using some values that were embedded in the kwik file during the initial run, and that it is not looking back into the prm file for the updated parameters I would like to use? It should look back to the prm on each call I guess since otherwise you cannot do different runs of KK with different parameters without re-doing everything incl. spike detection from the top, yeah?

This is the call I make:
$ phy cluster-auto params20150626_phy.prm

This is using phy on pyramidal, phy -v gives:
0.2.2.dev0-git-v0.2.1-200-g7f91b41c

KK2 reports:
KlustaKwik2 version git-0.2.4-1-g2dac4576

nsteinme, 2015-08-16:
I'm removing "done" unless you think I did something wrong in my test. Let me know, thanks -

rossant, 2015-08-16:
can you close this and open 2 new issues:

* kk2 log
* take params from PRM instead of KWIK in phy cluster-auto

thanks

nsteinme, 2015-08-16:
Sure

closed 2015-08-16 by nsteinme