cortex-lab/phy #562: phy cluster-auto myfile.prm doesn't work

nippoo, 2015-07-23:
```
(phy)nick@chandelier:~/data/Ehrlich/20150626$ phy cluster-auto /home/nick/data/Ehrlich/20150626/params20150626_phy.prm
Traceback (most recent call last):
  File "/apps/miniconda/envs/phy/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
  File "/apps/phy/phy/phy/scripts/phy_script.py", line 407, in main
    out = func(args)
  File "/apps/phy/phy/phy/scripts/phy_script.py", line 312, in cluster_auto
    session = _create_session(args, use_store=False)
  File "/apps/phy/phy/phy/scripts/phy_script.py", line 242, in _create_session
    session = Session(kwik_path, **kwargs)
  File "/apps/phy/phy/phy/session/session.py", line 84, in __init__
    gui_classes=self._gui_classes,
  File "/apps/phy/phy/phy/io/base.py", line 382, in __init__
    self.open(path, model=model)
  File "/apps/phy/phy/phy/session/session.py", line 128, in open
    return super(Session, self).open(model=model, path=kwik_path)
  File "/apps/phy/phy/phy/io/base.py", line 431, in open
    model = self._create_model(path)
  File "/apps/phy/phy/phy/session/session.py", line 100, in _create_model
    waveform_filter=self._waveform_filter,
  File "/apps/phy/phy/phy/io/kwik/model.py", line 276, in __init__
    clustering=clustering)
  File "/apps/phy/phy/phy/io/kwik/model.py", line 614, in open
    raise IOError("File `{0}` doesn't exist.".format(kwik_path))
OSError: File `/data/nick/Ehrlich/20150626/params20150626_phy.prm` doesn't exist.
```

Couple of issues here:

- `phy_script.cluster_auto()` should not `_create_session` since it's expecting a .prm - instead it should `_read_python(prm_file)` then `_create_session(kwik_path or params['experiment_name'] + '.kwik')`
- the `OSError` is misleading: https://github.com/kwikteam/phy/blob/master/phy/io/kwik/model.py#L613 checks if `self._kwik` is None, though in this case the filename exists but fails to be opened - `open_h5()` silently returns `None`, I guess... (?)

nippoo, 2015-07-23:
cc @nsteinme - will fix this soon

rossant, 2015-07-23:
you sure you're using the latest version?? this was fixed very recently...

nippoo, 2015-07-23:
on `master`:

```
def cluster_auto(args):
    assert args.file.endswith('.prm')
    session = _create_session(args, use_store=False)
    ns = dict(session=session,
              clustering=args.clustering,
              )
    cmd = ('session.cluster(clustering=clustering)')
    return (cmd, ns)
```

it's trying to `_create_session()` from a PRM, no?

rossant, 2015-07-23:
ahhh right there's something missing...

PR #573 (2015-07-27): [MRG] Fix cluster-auto from PRM file
Closes #562
closed 2015-07-27 by nippoo