cortex-lab/phy #564: probe file cannot be found?

nsteinme, 2015-07-24:
Trying to run phy spikesort on my local Windows machine, I get this:

```
[phy] N:\Ehrlich>phy spikesort params20150626_phy.prm
Traceback (most recent call last):
  File "C:\Anaconda\envs\phy\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line 384,
 in main
    out = func(args)
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line 313,
 in spikesort
    kwik_path=kwik_path)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\creator.py", line 344, in
 create_kwik
    probe = load_probe(params['prb_file'])
  File "c:\users\nick\documents\github\phy\phy\electrode\mea.py", line 84, in lo
ad_probe
    raise IOError("The probe `{}` cannot be found.".format(name))
OSError: The probe `N:\Ehrlich\imecToWhisper.prb` cannot be found.
```

But that file definitely does exist:
```
In [2]: os.path.isfile('N:\Ehrlich\imecToWhisper.prb')
Out[2]: True
```

rossant, 2015-07-24:
hmm is that the latest version?

rossant, 2015-07-24:
btw on Windows you may have problems with escape backslashes, and in a Python file you might need to prepend a `r` before the string, i.e.

```
probe = r'N:\Ehrlich\imecToWhisper.prb'
```

nsteinme, 2015-07-24:
Great, thanks - it was first that I was on pr/555, then I did have another error fixed by the 'r'. Works now.

closed 2015-07-24 by nsteinme