cortex-lab/phy #565: sampling an interval doesn't work if you don't start at zero

nsteinme, 2015-07-24:
Line 97 of detect/spikedetekt.py is a bug: it checks that start is less than n_samples, but if an interval is used, n_samples is the number of samples within the interval whereas start is in absolute samples within the whole file. E.g. if you ask for the interval 9,12 with sample_rate=20000, then start will be 180000 but nsamples will only be 60000, so the assertion will fail though the data and numbers are fine. 

Also can we clear the NotImplementedError, line 100? I would like to use this function and I don't much care whether the spike times are referenced to the start of the interval or to the start of the total file (in fact maybe the former is preferable?).

rossant, 2015-07-25:
good catch, it should be `assert 0 <= start < end`, PR welcome :)

rossant, 2015-07-25:
> Also can we clear the NotImplementedError, line 100? I would like to use this function and I don't much care whether the spike times are referenced to the start of the interval or to the start of the total file (in fact maybe the former is preferable?).

I'm not sure. @nippoo ?

nsteinme, 2015-07-25:
Yes, but so far posting an issue here is easier for me ;) I need to take
some time and get good at github...

rossant, 2015-07-25:
it's always good to post an issue before proposing any PR...
@nippoo we're counting on you to make @nsteinme a git expert

nsteinme, 2015-07-25:
Don't worry he's doing his best, I'm just being a slow learner ;)

nsteinme, 2015-07-27:
I will make a different issue about the NotImplementedError, since it needs to stay until that is implemented...

closed 2015-07-31 by rossant