cortex-lab/phy #575: addclustering.py

nsteinme, 2015-07-27:
To add a txt file containing a clustering to a kwik file.

rossant, 2015-07-27:
something like this should work, but it is untested

```python
# python addclustering.py myfile.kwik myclustering spikeclusters.txt

from phy.io.kwik import KwikModel

sc = np.loadtxt(sys.argv[3])
model = KwikModel(sys.argv[1])
model.add_clustering(sys.argv[2], sc)
```

nsteinme, 2015-07-27:
Will the result be written back to the kwik file permanently? That happens in model.add_clustering or model.close? If so I'll test it.

rossant, 2015-07-27:
Yes, that happens in `model.add_clustering()`. The `model.close()` is not stricly necessary actually (the file is reopened in write mode when adding the clustering, and closed immediately afterwards)

nsteinme, 2015-07-28:
In [7]: model.add_clustering('testClustering', sc)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-7-e2fb8150a16d> in <module>()
----> 1 model.add_clustering('testClustering', sc)

c:\users\nick\documents\github\phy\phy\io\kwik\model.py in add_clustering(self,
name, spike_clusters)
    796                            name,
    797                            channel_group=self._channel_group,
--> 798                            spike_clusters=spike_clusters,
    799                            )
    800

c:\users\nick\documents\github\phy\phy\io\kwik\model.py in _create_clustering(f,
 name, channel_group, spike_clusters, cluster_groups)
    137     f.write(path, spike_clusters.astype(np.int32))
    138
--> 139     cluster_ids = _unique(spike_clusters)
    140
    141     # Create cluster metadata.

c:\users\nick\documents\github\phy\phy\utils\array.py in _unique(x)
     68     x = _as_array(x)
     69     x = x[x >= 0]
---> 70     bc = np.bincount(x)
     71     return np.nonzero(bc)[0]
     72

TypeError: Cannot cast array data from dtype('float64') to dtype('int64') accord
ing to the rule 'safe'

nsteinme, 2015-07-28:
got it:  np.int64(sc)

nsteinme, 2015-07-28:
Just to confirm, this function seems to work, the added clustering loaded successfully and looks correct. Is this a function you want added to the repository?

rossant, 2015-07-28:
ok use `np.loadtxt(sys.argv[3], dtype=np.int64)` then

rossant, 2015-07-28:
> Just to confirm, this function seems to work, the added clustering loaded successfully and looks correct. Is this a function you want added to the repository?

I'm not sure, this will all be obsolete when the file format changes again... (and anyway the function is already part of the library, it's called `add_clustering()`...)

closed 2015-07-31 by rossant