cortex-lab/phy #581: Can't switch shank in manual clustering

tymoreau, 2015-07-28:
I ran phy on a dataset and tried to use the GUI for manual clustering. For the first shank, it works fine, but I can't access the other three.  
Here's what I do to open shank 1:

    import phy
    phy.enable_qt()
    from phy.session import Session
    kwik_path="Rat034_2015_02_16_12_51.kwik"
    session=Session(kwik_path)
    session.show_gui()

Then I close the GUI and try to open shank 2 with `session.change_channel_group(2)`. I get the following error:

    10:33:05 [I] Switched to channel group 2.
    Features and masks initialized.
    ---------------------------------------------------------------------------
    [...]
    /home/david/phy/phy/io/kwik/store_items.py in unmasked_channels(self, cluster)
        587     def unmasked_channels(self, cluster):
        588         mean_masks = self.load(cluster, 'mean_masks')
    --> 589         return np.nonzero(mean_masks > .1)[0]
        590 
        591     def n_unmasked_channels(self, cluster):
    TypeError: unorderable types: NoneType() > float()

For the shanks 3 and 4, the error is different:

    10:37:01 [I] Switched to channel group 3.
    ---------------------------------------------------------------------------
    [...]
    /home/david/phy/phy/io/kwik/store_items.py in _store(self, cluster, chunk_spikes, chunk_spikes_per_cluster, chunk_features_masks)
         96         # Extract features and masks for that cluster, in the
         97         # current chunk.
    ---> 98         tmp = chunk_features_masks[idx, :]
         99 
        100         # NOTE: channel order has already been taken into account

    IndexError: index 37195 is out of bounds for axis 0 with size 5928

You can find the kwik, prm and prb files on [google drive](https://drive.google.com/folderview?id=0B3Ae5d7CRAdYfktvS3JDLVROS3M2TGVQUXdYc1hCWVBKVUFtcHFCV1JZVzBvcGJad2w1Qzg&usp=sharing)

rossant, 2015-07-28:
Hi,

* Could you upload the `.kwx` as well, or is it too big?
* What program did you use for spike detection and automatic clustering (klusta or phy)?

rossant, 2015-07-31:
@tymoreau does it work with `session=Session(kwik_path, channel_group=2)`?

ablot, 2015-08-01:
It doesn't work for me. 

Using phy `0.2.2.dev0`, and `session=Session(kwik_path, channel_group=1)`  seems OK at first, but the output of `session.model.describe()` says there is 0 Clusters and `session.store.display_status()` raises an error:

`session.model.describe()`
```
Kwik file               longpathremoved
Recordings              9
List of shanks          0, 1*
Clusterings             
Channels                30
Spikes                  232790
Clusters                0
Duration                3995s
```
`session.store.display_status()`
```python
phy/phy/io/store.py in status(self)
    599     def status(self):
    600         """Return the current status of the cluster store."""
--> 601         in_store = set(self.disk_store.cluster_ids)
    602         valid = set(self.cluster_ids)
    603         invalid = in_store - valid

AttributeError: 'NoneType' object has no attribute 'cluster_ids'
```

It looks as if the second shank had not been clustered. But I think `phy spikesort` worked properly. There was no error message and there are multiple clusters in the folder `AF_59_150727_all.phy/cluster_store/1/main/`.

rossant, 2015-08-01:
does it work if you use `phy cluster-manual file.kwik --channel-group=XXX` in a terminal rather than in IPython?

tymoreau, 2015-08-03:
I used phy to do the spike detection and clustering. I've uploaded the .kwx on Drive.

I've just installed the last development version of phy.
With the command line  ` phy cluster-manual file.kwik --channel-group=2`, I get:

      File "/home/david/anaconda/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.2.dev0', 'console_scripts', 'phy')()
       File "/home/david/phy/phy/scripts/phy_script.py", line 427, in main
    out = func(args)
       File "/home/david/phy/phy/scripts/phy_script.py", line 368, in cluster_manual
    use_store=not(args.no_store),
       File "/home/david/phy/phy/scripts/phy_script.py", line 247, in _create_session
    session = Session(kwik_path, **kwargs)
       File "/home/david/phy/phy/session/session.py", line 86, in __init__
    gui_classes=self._gui_classes,
       File "/home/david/phy/phy/io/base.py", line 382, in __init__
    self.open(path, model=model)
       File "/home/david/phy/phy/session/session.py", line 131, in open
    return super(Session, self).open(model=model, path=kwik_path)
       File "/home/david/phy/phy/io/base.py", line 431, in open
    model = self._create_model(path)
       File "/home/david/phy/phy/session/session.py", line 103, in _create_model
    waveform_filter=self._waveform_filter,
       File "/home/david/phy/phy/io/kwik/model.py", line 279, in __init__
    clustering=clustering)
       File "/home/david/phy/phy/io/kwik/model.py", line 659, in open
    self._clustering_changed(self._clustering)
       File "/home/david/phy/phy/io/kwik/model.py", line 728, in _clustering_changed
    raise ValueError("The clustering {0} is invalid.".format(value))
     ValueError: The clustering main is invalid.

On IPython, ` session=Session(kwik_path, channel_group=2)` prints nothing, and then `session.show_gui()` gives me:

```
    [...]
    /home/david/phy/phy/cluster/manual/gui.py in on_open(self)
    377         self._create_global_history()
    378         # This connects the callback that updates the model spikes_per_cluster.
--> 379         self._create_clustering()
    380         self._create_cluster_metadata()
    381         # This connects the callback that updates the store.

/home/david/phy/phy/cluster/manual/gui.py in _create_clustering(self)
    285 
    286     def _create_clustering(self):
--> 287         self.clustering = Clustering(self.model.spike_clusters)
    288 
    289         @self.connect

/home/david/phy/phy/cluster/manual/clustering.py in __init__(self, spike_clusters)
    157         self._undo_stack = History(base_item=(None, None))
    158         # Spike -> cluster mapping.
--> 159         self._spike_clusters = _as_array(spike_clusters)
    160         self._n_spikes = len(self._spike_clusters)
    161         self._spike_ids = np.arange(self._n_spikes).astype(np.int64)

/home/david/phy/phy/utils/_types.py in _as_array(arr, dtype)
     81     if out.dtype not in _ACCEPTED_ARRAY_DTYPES:
     82         raise ValueError("'arr' seems to have an invalid dtype: "
---> 83                          "{0:s}".format(str(out.dtype)))
     84     return out
     85 

ValueError: 'arr' seems to have an invalid dtype: object
```

rossant, 2015-08-03:
looks like a bug where klustakwik is only run on the first shank. that's why the clustering is said to be invalid: it doesn't exist!

in the meantime you might want to try `session = Session(file, channel_group=XXX); session.cluster()`

ibadr, 2015-08-16:
I think this bug is not completely fixed yet, even with #608 already merged. While using IPython, I got the following traceback upon switching shanks (same settings as in #619. 

```
In [9]: session.change_channel_group(1)
14:05:37 [I] Switched to channel group 1.
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-9-c17075416b5f> in <module>()
----> 1 session.change_channel_group(1)

c:\users\islam\phy\phy\session\session.py in change_channel_group(self, channel_
group)
    207         self.model.channel_group = channel_group
    208         info("Switched to channel group {}.".format(channel_group))
--> 209         self.emit('open')
    210
    211     def change_clustering(self, clustering):

c:\users\islam\phy\phy\utils\event.py in emit(self, event, *args, **kwargs)
    126                 kwargs = {n: v for n, v in kwargs.items()
    127                           if n in argspec.args}
--> 128             res.append(callback(*args, **kwargs))
    129         return res
    130

c:\users\islam\phy\phy\session\session.py in on_open(self)
    217
    218     def on_open(self):
--> 219         self._create_cluster_store()
    220
    221     def on_close(self):

c:\users\islam\phy\phy\session\session.py in _create_cluster_store(self)
    201         # If the cluster store already exists and is consistent
    202         # with the data, it is not recreated.
--> 203         self.store.generate()
    204
    205     def change_channel_group(self, channel_group):

c:\users\islam\phy\phy\io\store.py in generate(self, mode)
    661         debug("Initializing the cluster store for {0:s}.".format(name))
    662         for item in self._items.values():
--> 663             item.store_all(mode)
    664
    665     # Load

c:\users\islam\phy\phy\io\kwik\store_items.py in store_all(self, mode)
    211                                 chunk_spikes,
    212                                 chunk_spc,
--> 213                                 chunk_features_masks,
    214                                 )
    215

c:\users\islam\phy\phy\io\kwik\store_items.py in _store(self, cluster, chunk_spi
kes, chunk_spikes_per_cluster, chunk_features_masks)
     96         # Extract features and masks for that cluster, in the
     97         # current chunk.
---> 98         tmp = chunk_features_masks[idx, :]
     99
    100         # NOTE: channel order has already been taken into account

IndexError: index 98217 is out of bounds for axis 0 with size 81276
```

PR #667 (2015-10-08): [MRG] fix #581 and fix #620
* Fixes #581 
* Fixes #620

Note that when showing a GUI with a new clustering/shank you need to use `session.show_gui()` to create and load a new GUI rather than `gui.show()` which will try and show the old one with disastrous results...
commit 5026798d (2015-10-08): Merge pull request #667 from nippoo/switch_clusterings

[MRG] fix #581 and fix #620
commit 9e2e8c38 (2015-10-08): fix #581 and fix #620
closed 2015-10-08 by rossant