cortex-lab/phy #583: memory error

nsteinme, 2015-07-28:
Error text below. As you can see there are 2.5 million spikes in this file, not an extraordinary number. Maybe the problem is that there is just one cluster? I have never opened a kwik file with just one cluster before. Another file opens without error. 

[phy] N:\Ehrlich>phy cluster-manual 20150707_all.kwik
Kwik file               N:\Ehrlich\20150707_all.kwik
Recordings              1
List of shanks          1*
Clusterings             main*, empty
Channels                120
Spikes                  2465090
Clusters                1
Duration                0s
Traceback (most recent call last):
  File "C:\Anaconda\envs\phy\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.0.dev0', 'console_scripts', 'phy')()
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line 422,
 in main
    out = func(args)
  File "c:\users\nick\documents\github\phy\phy\scripts\phy_script.py", line 373,
 in cluster_manual
    gui = session.show_gui(cluster_ids=cluster_ids, show=False)
  File "c:\users\nick\documents\github\phy\phy\session\session.py", line 450, in
 show_gui
    **kwargs)
  File "c:\users\nick\documents\github\phy\phy\io\base.py", line 468, in show_gu
i
    **params)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 248, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 217, in _creat
e_widget
    return widget_class(**kwargs)
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\gui.py", line 107,
 in __init__
    **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 353, in __init
__
    self._initialize_views()
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\gui.py", line 125,
 in _initialize_views
    super(ClusterManualGUI, self)._initialize_views()
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 361, in _initi
alize_views
    requested_count=self._state.get('view_count', None),
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 433, in _load_
config
    self.add_view(name, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 597, in add_vi
ew
    item = self._view_creator.add(item, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 248, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 217, in _creat
e_widget
    return widget_class(**kwargs)
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\view_models.py", l
ine 64, in __init__
    self.select(_as_list(cluster_ids))
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\view_models.py", l
ine 93, in select
    self.on_select(cluster_ids, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\view_models.py", l
ine 139, in on_select
    self.update(cluster_ids=cluster_ids)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 180, in update

    self._update(self._view, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\gui\base.py", line 158, in _updat
e
    html = self.get_html(**kwargs)
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\view_models.py", l
ine 265, in get_html
    np.ndarray) for name in names}
  File "c:\users\nick\documents\github\phy\phy\cluster\manual\view_models.py", l
ine 265, in <dictcomp>
    np.ndarray) for name in names}
  File "c:\users\nick\documents\github\phy\phy\io\store.py", line 526, in load
    return self.load(name, *args, **kwargs)
  File "c:\users\nick\documents\github\phy\phy\io\store.py", line 675, in load
    return item.load(clusters, name)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\store_items.py", line 670
, in load
    return getattr(self, name)(cluster)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\store_items.py", line 585
, in mean_waveforms
    return self._load_mean(cluster, 'mean_waveforms')
  File "c:\users\nick\documents\github\phy\phy\io\kwik\store_items.py", line 576
, in _load_mean
    return getattr(item, name)(cluster)
  File "c:\users\nick\documents\github\phy\phy\io\kwik\store_items.py", line 421
, in mean_waveforms
    return self.waveforms_and_mean(cluster)[1]
  File "c:\users\nick\documents\github\phy\phy\io\kwik\store_items.py", line 416
, in waveforms_and_mean
    waveforms = self.model.waveforms[spikes].astype(np.float32)
MemoryError

rossant, 2015-07-28:
where can I find the file on the server?

nsteinme, 2015-07-28:
/home/nick/data/Ehrlich/20150707_all.kwik

Ok, the memory error may have happened because I (yet again) forgot to
bring the raw data file to my local folder along with the kwik and kwx
files. But now that it is there, I see a different bug - the program just
takes forever (after describing itself but before "initializing features
and masks") slowly putting something into RAM. It got up to 22GB RAM usage
before I killed it (and that took probably 20 minutes or more) - the raw
file itself is only 19GB.

nsteinme, 2015-07-28:
Ok, I think it is likely to be something about the 1-cluster with 2.5million spikes, because I added another clustering and now things load just fine.

rossant, 2015-07-28:
Ok, reading your description, it looks like a h5py problem where loading many rows is extremely slow. This will be fixed when we get rid of HDF5.

nsteinme, 2015-07-28:
What is it reading many rows of? Why does that account for filling RAM?

I guess a hack solution in the meantime, if I want to look at a file before
any automatic clustering has been done, might be to "add clustering" with a
bunch of random integers since having a clustering seems to clear it up.

rossant, 2015-07-28:
`features_masks`: it should be loaded chunk by chunk, but I guess that for some reason the n_clusters=1 degenerate case is highly inefficient because of h5py

closed 2016-03-26 by rossant