cortex-lab/phy #588: GUI does not launch (running on server)

shabnamkadir, 2015-07-29:
```
In [1]:

import phy

phy.enable_qt()

from phy.session import Session

​

21:55:09 [I] Qt event loop activated.

In [2]:

kwik_path = '/home/skadir/globalphy/nicktest/20141202_child_320001.kwik'

In [3]:

session = Session(kwik_path)

Features and masks initialized.
Initializing statistics: 100.0%.
Statistics initialized.
Initializing statistics: 100.0%.

In [4]:

session.show_gui()

Out[4]:

<phy.cluster.manual.gui.ClusterManualGUI at 0x7f54b549de10>

WARNING: b'X Error: GLXBadRenderRequest 178\n  Extension:    156 (Uknown extension)\n  Minor opcode: 1 (Unknown request)\n  Resource id:  0x2'
WARNING:vispy:b'X Error: GLXBadRenderRequest 178\n  Extension:    156 (Uknown extension)\n  Minor opcode: 1 (Unknown request)\n  Resource id:  0x2'
WARNING: b'X Error: BadRequest (invalid request code or no such operation) 1\n  Extension:    156 (Uknown extension)\n  Minor opcode: 187 (Unknown request)\n  Resource id:  0x2'
WARNING:vispy:b'X Error: BadRequest (invalid request code or no such operation) 1\n  Extension:    156 (Uknown extension)\n  Minor opcode: 187 (Unknown request)\n  Resource id:  0x2'
```

shabnamkadir, 2015-07-29:
This might be helpful:

```
ms referer=None
Xlib:  extension "NV-GLX" missing on display "localhost:17.0".
Xlib:  extension "NV-GLX" missing on display "localhost:17.0".
Xlib:  extension "NV-GLX" missing on display "localhost:17.0".
Xlib:  extension "NV-GLX" missing on display "localhost:17.0".
Xlib:  extension "NV-GLX" missing on display "localhost:17.0".
[xcb] Too much data requested from _XRead
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
python3: ../../src/xcb_io.c:736: _XRead: Assertion `!xcb_xlib_too_much_data_requested' failed.

```

rossant, 2015-07-29:
The GUI cannot work remotely, unfortunately

closed 2016-04-06 by rossant