cortex-lab/phy #589: standalone traceview

nsteinme, 2015-07-30:
There was another issue about traceview improvements, but I think this is independent - 

Could there be a way to open up just the traceview from a .dat file directly? 

And as a really useful extension, could it try to perform spike detection on just the segment you're looking at and highlight the spikes? It could be a smaller/simpler version of detection, since you just need the masks really.  

As an extension on that, snippets to change the detection parameters, with immediate re-detection of the viewed segment? 

Ideally it would incorporate a way to jump around the file, e.g. with a snippet.

nippoo, 2015-07-30:
Duplicate of #459. We're on it! :)

nsteinme, 2015-07-30:
k thanks - added the additional suggestions to that thread; closing this one.

closed 2015-07-30 by nsteinme