cortex-lab/phy #591: Clustering several dat files / opening Klustaviewa files

AntonioFR8, 2015-07-30:
In a linux sever (SUSE) I have been able of installing the phy and running it for the test data and also for one small sample of our own data. However I can't open any of the kwik files that we have clustered before with KlustaViewa although it properly initialized masks, features and statistics (please see attached a text file with the terminal message). I was wondering if the problem may be related with the fact that all the sessions we clustered, and are trying now to open with phy, were composed by several separated dat files. When I make several copies of your hybrid_10s.dat file, rename them and try to cluster with phy it get stucked in the Finding the thresholds step. However if I previously concatenated those several copies of the original dat file, everything works without problem. Is there any problem with adding several dat files in the .prm file and cluster them together?

nippoo, 2015-07-30:
Hi, I can't see the text file with the message?

AntonioFR8, 2015-07-30:
Sorry, here it is.

azahara@tectum:~/testdata/AYA3_150503> phy cluster-manual AYA3_150503.kwik
Features and masks initialized.
Statistics initialized.
Kwik file               /home/azahara/testdata/AYA3_150503/AYA3_150503.kwik
Recordings              3
List of shanks          2*
Clusterings             main*, original
Channels                30
Spikes                  1807520
Clusters                201
Duration                0s
Traceback (most recent call last):
  File "/home/azahara/miniconda/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.1', 'console_scripts', 'phy')()
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/scripts/phy_script.py", line 391, in main
    out = func(args)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/scripts/phy_script.py", line 347, in cluster_manual
    gui = session.show_gui(cluster_ids=cluster_ids, show=False)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/session.py", line 438, in show_gui
    **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/io/base.py", line 468, in show_gui
    **params)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 247, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 216, in _create_widget
    return widget_class(**kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/manual/gui.py", line 107, in __init__
    **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 352, in __init__
    self._initialize_views()
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/manual/gui.py", line 125, in _initialize_views
    super(ClusterManualGUI, self)._initialize_views()
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 360, in _initialize_views
    requested_count=self._state.get('view_count', None),
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 432, in _load_config
    self.add_view(name, **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 593, in add_view
    item = self._view_creator.add(item, **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 247, in add
    widget = self._create_widget(widget_class, **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 216, in _create_widget
    return widget_class(**kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 571, in __init__
    super(TraceViewModel, self).__init__(**kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 158, in __init__
    super(VispyViewModel, self).__init__(**kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 60, in __init__
    **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/gui/base.py", line 50, in __init__
    self.on_open()
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 702, in on_open
    self.select([])
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 222, in select
    self.on_select(cluster_ids, **kwargs)
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 727, in on_select
    self.interval = sample - half_size, sample + half_size
  File "/home/azahara/miniconda/lib/python3.4/site-packages/phy/cluster/view_models.py", line 641, in interval
    assert 0 <= start < end <= n
AssertionError

Antonio

rossant, 2015-09-03:
can you send us your data so that we can properly debug?

AntonioFR8, 2015-09-04:
Hi,

Which data are you looking for? The problem of not running separate dat files I got it just with your hybrid_10s.dat test file. For the problem to open the files previously clustered with KlustaViewa I can send you some of my files, although are going to be heavy.

Thanks,
Antonio FR

rossant, 2015-09-04:
closing in favor of #634 -- this will be fixed soon

closed 2015-09-04 by rossant