cortex-lab/phy #598: error when attempting to view 15000 clusters without the cluster store

shabnamkadir, 2015-07-31:
The perversity of what I am attempting withstanding:
```

Cell Toolbar:
In [1]:

import phy

phy.enable_qt()

from phy.session import Session

14:15:59 [I] Qt event loop activated.

In [2]:

import numpy as np

In [3]:

kwik_path = '/mnt/nfs/homedirs/skadir/globalphy/nicktest/20141202_child_320001.kwik'

In [5]:

session = Session(kwik_path, use_store=False)

In [6]:

session.change_clustering('clust2')

14:16:54 [I] Switched to `clust2` clustering.

In [ ]:

Session(self, kwik_path=None, clustering=None, channel_group=None, model=None, use_store=False, phy_user_dir=None, waveform_filter=True)

In [7]:

gui = session.show_gui()

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-7-250ec0f8868a> in <module>()
----> 1 gui = session.show_gui()

/home/skadir/globalphy/phy/phy/session/session.py in show_gui(self, **kwargs)
    451         """Show a GUI."""
    452         gui = super(Session, self).show_gui(store=self.store,
--> 453                                             **kwargs)
    454 
    455         @gui.connect

/home/skadir/globalphy/phy/phy/io/base.py in show_gui(self, name, show, **kwargs)
    466                                     model=self.model,
    467                                     settings=self.settings,
--> 468                                     **params)
    469         gui._save_state = True
    470 

/home/skadir/globalphy/phy/phy/gui/base.py in add(self, widget_class, show, **kwargs)
    246                                  "`{}`.".format(widget_class))
    247             widget_class = self.widget_classes[widget_class]
--> 248         widget = self._create_widget(widget_class, **kwargs)
    249         if widget not in self._widgets:
    250             self._widgets.append(widget)

/home/skadir/globalphy/phy/phy/gui/base.py in _create_widget(self, widget_class, **kwargs)
    215 
    216         """
--> 217         return widget_class(**kwargs)
    218 
    219     @property

/home/skadir/globalphy/phy/phy/cluster/manual/gui.py in __init__(self, model, store, cluster_ids, **kwargs)
    105         super(ClusterManualGUI, self).__init__(model=model,
    106                                                vm_classes=self._vm_classes,
--> 107                                                **kwargs)
    108 
    109     def _initialize_views(self):

/home/skadir/globalphy/phy/phy/gui/base.py in __init__(self, model, vm_classes, state, shortcuts, snippets, config, settings)
    351         self._dock = DockWindow(title=self.title)
    352         self._view_creator = WidgetCreator(widget_classes=vm_classes)
--> 353         self._initialize_views()
    354         self._load_geometry_state(state)
    355         self._set_default_shortcuts()

/home/skadir/globalphy/phy/phy/cluster/manual/gui.py in _initialize_views(self)
    120 
    121         self.on_open()
--> 122         self.wizard.start()
    123         if self._cluster_ids is None:
    124             self._cluster_ids = self.wizard.selection

/home/skadir/globalphy/phy/phy/cluster/manual/wizard.py in start(self)
    369     def start(self):
    370         """Start the wizard by setting the list of best clusters."""
--> 371         self._set_best_list()
    372 
    373     def pin(self, cluster=None):

/home/skadir/globalphy/phy/phy/cluster/manual/wizard.py in _set_best_list(self, cluster, clusters)
    220             cluster = self.best
    221         if clusters is None:
--> 222             clusters = self.best_clusters()
    223         self._best_list = clusters
    224         if clusters:

/home/skadir/globalphy/phy/phy/cluster/manual/wizard.py in best_clusters(self, n_max, quality)
    192         if quality is None:
    193             quality = self._quality
--> 194         best = _best_clusters(self.cluster_ids, quality, n_max=n_max)
    195         return self._sort(best)
    196 

/home/skadir/globalphy/phy/phy/cluster/manual/wizard.py in _best_clusters(clusters, quality, n_max)
     33 def _best_clusters(clusters, quality, n_max=None):
     34     return _argsort([(cluster, quality(cluster))
---> 35                      for cluster in clusters], n_max=n_max)
     36 
     37 

/home/skadir/globalphy/phy/phy/cluster/manual/wizard.py in <listcomp>(.0)
     33 def _best_clusters(clusters, quality, n_max=None):
     34     return _argsort([(cluster, quality(cluster))
---> 35                      for cluster in clusters], n_max=n_max)
     36 
     37 

/home/skadir/globalphy/phy/phy/cluster/manual/gui.py in quality(cluster)
    332             """Return the maximum mean_masks across all channels
    333             for a given cluster."""
--> 334             return self.store.mean_masks(cluster).max()
    335 
    336         @self.connect

/home/skadir/globalphy/phy/phy/io/store.py in load(*args, **kwargs)
    524         def _make_func(name):
    525             def load(*args, **kwargs):
--> 526                 return self.load(name, *args, **kwargs)
    527             return load
    528 

/home/skadir/globalphy/phy/phy/io/store.py in load(self, name, clusters, spikes)
    673             if _is_integer(clusters):
    674                 # Single cluster case.
--> 675                 return item.load(clusters, name)
    676             clusters = np.unique(clusters)
    677             if spikes is None:

/home/skadir/globalphy/phy/phy/io/kwik/store_items.py in load(self, cluster, name)
    668             # the corresponding method.
    669             if hasattr(self, name):
--> 670                 return getattr(self, name)(cluster)
    671             # Custom statistic.
    672             else:

/home/skadir/globalphy/phy/phy/io/kwik/store_items.py in mean_masks(self, cluster)
    577 
    578     def mean_masks(self, cluster):
--> 579         return self._load_mean(cluster, 'mean_masks')
    580 
    581     def mean_features(self, cluster):

/home/skadir/globalphy/phy/phy/io/kwik/store_items.py in _load_mean(self, cluster, name)
    574                          else 'features and masks')
    575             item = self.cluster_store.items[item_name]
--> 576             return getattr(item, name)(cluster)
    577 
    578     def mean_masks(self, cluster):

/home/skadir/globalphy/phy/phy/io/kwik/store_items.py in mean_masks(self, cluster)
    121 
    122     def mean_masks(self, cluster):
--> 123         masks = self.cluster_store.masks(cluster)
    124         mean_masks = _mean(masks, (self.n_channels,))
    125         return mean_masks

/home/skadir/globalphy/phy/phy/io/store.py in load(*args, **kwargs)
    524         def _make_func(name):
    525             def load(*args, **kwargs):
--> 526                 return self.load(name, *args, **kwargs)
    527             return load
    528 

/home/skadir/globalphy/phy/phy/io/store.py in load(self, name, clusters, spikes)
    673             if _is_integer(clusters):
    674                 # Single cluster case.
--> 675                 return item.load(clusters, name)
    676             clusters = np.unique(clusters)
    677             if spikes is None:

/home/skadir/globalphy/phy/phy/io/kwik/store_items.py in load(self, cluster, name)
    242         # Fallback to load_spikes if the data could not be obtained from
    243         # the store.
--> 244         spikes = self.spikes_per_cluster[cluster]
    245         return self.load_spikes(spikes, name)
    246 

KeyError: 228
```

shabnamkadir, 2015-07-31:
Totally uninformed conjecture: I am opening an .kwik file with 2 clsuterings, one has 227 clusters, the other around 15000 clusters. the clustering with 227 clusters has a 'cluster store', whereas the 15000 cluster clustering doesn't. The emergence of the number 228, may have something to do with this, I imagine.

rossant, 2015-07-31:
can you please try `Session(kwik_path, use_store=False, clustering='clust2')`

shabnamkadir, 2015-07-31:
OK, that worked!

rossant, 2015-07-31:
ok so the error is unrelated to the fact that you have 15000 clusters, but that you changed the clustering after the session was opened (which is a totally valid use-case).

rossant, 2016-03-26:
fixed

closed 2016-03-26 by rossant