cortex-lab/phy #616: TypeError trying to run cluser-manual

CamHiggins, 2015-08-12:
Hi there,

I'm unable to run the manual clustering step - I can run the spike detect and auto clustering code fine, but as soon as I try to open cluster-manual it crashes, with the below error message. Any tips?

Thanks.

(klusta)126 chiggins@jalapeno00 $ phy cluster-manual Rat2_32Chs_Wired_20150130_Test1.kwik
Features and masks initialized.
Traceback (most recent call last):
  File "/home/fs0/chiggins/.conda/envs/klusta/bin/phy", line 9, in <module>
    load_entry_point('phy==0.2.0', 'console_scripts', 'phy')()
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/scripts/phy_script.py", line 391, in main
    out = func(args)
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/scripts/phy_script.py", line 337, in cluster_manual
    use_store=not(args.no_store),
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/scripts/phy_script.py", line 237, in _create_session
    session = Session(kwik_path, **kwargs)
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/cluster/session.py", line 85, in __init__
    gui_classes=self._gui_classes,
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/io/base.py", line 382, in __init__
    self.open(path, model=model)
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/cluster/session.py", line 129, in open
    return super(Session, self).open(model=model, path=kwik_path)
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/io/base.py", line 435, in open
    self.emit('open')
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/utils/event.py", line 128, in emit
    res.append(callback(*args, **kwargs))
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/cluster/session.py", line 217, in on_open
    self._create_cluster_store()
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/cluster/session.py", line 201, in _create_cluster_store
    self.store.generate()
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/io/store.py", line 663, in generate
    item.store_all(mode)
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/io/kwik/store_items.py", line 458, in store_all
    chunk_waveforms = self.model.waveforms[spk]
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/traces/waveform.py", line 376, in __getitem__
    return self._waveforms[times]
  File "/home/fs0/chiggins/.conda/envs/klusta/lib/python2.7/site-packages/phy/traces/waveform.py", line 356, in __getitem__
    waveforms -= self._dc_offset
TypeError: unsupported operand type(s) for -=: 'numpy.ndarray' and 'unicode'
(klusta)127 chiggins@jalapeno00 $

rossant, 2015-08-20:
Hi, could you send us the files? (kwik, kwx, prm, dat, etc.)

CamHiggins, 2015-08-25:
Dear Cyrille,

Thanks for getting back to me. I have put the files up on our university file sharing service – you should receive an email with the download instructions to this email address (let me know if there is another email address to give access to). Please let me know once you have downloaded them so I can take them down – also these files are not for further use or distribution other than debugging please.

Let me know if there is anything else I can do or send to you – thanks again for your help.

Cameron

rossant, 2015-08-25:
I haven't received any e-mail yet. Have you used my gmail address?

CamHiggins, 2015-08-26:
No, I sent it to this email address. If it hasn’t come through, could you let me know what address to send it to? Thanks.

rossant, 2015-08-26:
What is "this" e-mail address? I'm receiving this on GitHub (https://github.com/kwikteam/phy/issues/616#issuecomment-134617893) and on my gmail address firstname.lastname@gmail.com.

CamHiggins, 2015-08-26:
Ok – have just entered your gmail address, I think there is a delay but you should receive the invitation to download the files soon to your gmail. Thanks very much!

rossant, 2015-08-26:
got all of it, thanks!

rossant, 2015-08-26:
OK so your dataset works fine for me. I have a folder with only the following files:

```
-rw-r-----@ 1 cyrille  staff   2,5K 26 aoû 10:37 32chs2shanksParallel.prb
-rw-r-----@ 1 cyrille  staff   707M 26 aoû 10:37 Rat2_32Chs_Wired_20150130_Test1.dat
-rw-r-----@ 1 cyrille  staff   2,6G 26 aoû 10:38 Rat2_32Chs_Wired_20150130_Test1.kwik
-rw-r-----@ 1 cyrille  staff    92M 26 aoû 10:37 Rat2_32Chs_Wired_20150130_Test1.kwx
-rw-r-----@ 1 cyrille  staff   1,0K 26 aoû 10:37 rat2.prm
```

I ran `phy cluster-manual Rat2_32Chs_Wired_20150130_Test1.kwik`. The cluster store is created and then the GUI opens with no error. I have the latest version of phy (development version) that you can install with `git clone git://github.com/kwikteam/phy.git ~/phy && cd ~/phy && python setup.py develop`.

I notice that you use the `klusta` environment; however this is the environment for the old version of the suite, not phy. You should deactivate this environment before running phy. Have you created an environment for phy? You'll find more information at http://phy.readthedocs.org/en/latest/install/

CamHiggins, 2015-08-26:
Ah ok – I will try downloading the latest version of phy and see how that does. I thought the python environments were the same version, so will check that too – thanks.

rossant, 2015-09-03:
@CamHiggins did it work?

closed 2015-09-09 by rossant