cortex-lab/phy #618: model.close() is not called on exit

nippoo, 2015-08-14:
If the GUI is closed with alt+f4 or the close button, the file handles aren't properly closed.

nippoo, 2015-08-14:
... or maybe they are. Trying to call `self.close()` from `self.__del__` results in nasty warnings like `C:\Users\Max\miniconda3\lib\site-packages\h5py\_hl\files.py:290: RuntimeWarning: Found murdered identifier 112489032 of kind <class 'NoneType'> HDF5 id 1376727968`...

rossant, 2015-08-14:
murdered? omg

nippoo, 2015-08-14:
yeah. thought this would be a simple bug. I'm actually not sure it matters. I think all the files will individually be closed on exit... so it might not be a problem

rossant, 2015-08-14:
there are segfaults when quitting sometimes, so something might be wrong...

closed 2016-03-24 by rossant