cortex-lab/phy #619: Assertion fails with single-channel data in _dimensions(x_channels, y_channels) in view_models.py

ibadr, 2015-08-14:
I'm on the dev branch (SHA1: 7f91b41c7b86df2fc3a4716fc80223851ae130d2), which I cloned using `git clone git://github.com/kwikteam/phy.git`. I'm trying to sort 32-channel data from TDT microwire arrays where I treated every channel as a single shank using:
```
channel_groups = {
    0: { "channels":[0], "graph":[[0,0],], },
    1: { "channels":[1], "graph":[[1,1],] },
...
```

However, I ran into an issue while viewing the sorting results, where I had to add this line to `phy/cluster/manual/view_models.py` beginning line 762:

```
     n = len(x_channels)
+    if not n: return 0, 0
     assert len(y_channels) == n
```

I had to do this because the code was crashing at the assertion because `len(x_channels)==0` whereas `y_channels` was of length one. The GUI code works fine after this edit except for a warning about division by zero, and the scale of most panels is messed up and I have to manually correct it. So, I was wondering what is going on and how to fix this issue?

rossant, 2015-08-20:
have you tried using a single shank with an empty graph?

snaki001, 2015-10-23:
Hi guys,

I have a similar issue when running manual clustering. For me, the problem occurs at 'assert len(y_channels) == n' in view_models.py
I'm using 1 shank for 4 channels.
spikesort runs fine.

Do you have any idea how to fix this ?

Thanks

NikiGuo, 2015-11-26:
I got a similar error when running 'phy cluster-manual myfile.kwik' : line 765, in_dimensions assert len(y_channels)==n  AssertionError  ,I'm using 8 shank for 4 channels .

NikiGuo, 2015-11-30:
I think the features per channels should be set lower when I have low channel counts in one shank.

rossant, 2016-03-30:
should be fixed in the latest version

closed 2016-03-30 by rossant