cortex-lab/phy #620: Attempting to re-sort a single shank (channel-group) results in out-of-bounds crash

ibadr, 2015-08-14:
Same settings as in #619 .. now when trying to re-sort data on a single shank, I get this traceback

```
$ phy cluster-auto --channel-group=2 tdt.prm
Features and masks initialized.←[K
Initializing statistics: 100.0%.
Statistics initialized.←[K
Initializing statistics: 100.0%.
10:32:06 [I] Switched to channel group 2.
Traceback (most recent call last):0.0%.
  File "c:\Users\islam\Miniconda3\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.2.dev0', 'console_scripts', 'phy')()
  File "c:\users\islam\phy\phy\scripts\phy_script.py", line 460, in main
    exec_(cmd, {}, ns)
  File "<string>", line 1, in <module>
  File "c:\users\islam\phy\phy\session\session.py", line 375, in cluster
    self.change_channel_group(channel_group)
  File "c:\users\islam\phy\phy\session\session.py", line 209, in change_channel_
group
    self.emit('open')
  File "c:\users\islam\phy\phy\utils\event.py", line 128, in emit
    res.append(callback(*args, **kwargs))
  File "c:\users\islam\phy\phy\session\session.py", line 219, in on_open
    self._create_cluster_store()
  File "c:\users\islam\phy\phy\session\session.py", line 203, in _create_cluster
_store
    self.store.generate()
  File "c:\users\islam\phy\phy\io\store.py", line 663, in generate
    item.store_all(mode)
  File "c:\users\islam\phy\phy\io\kwik\store_items.py", line 213, in store_all
    chunk_features_masks,
  File "c:\users\islam\phy\phy\io\kwik\store_items.py", line 98, in _store
    tmp = chunk_features_masks[idx, :]
IndexError: index 61289 is out of bounds for axis 0 with size 59737
```

This is my parameter file
```
experiment_name = 'experiment_Block-2'
prb_file = '32chanTDT-PFC.prb'

traces = dict(
    raw_data_files=[experiment_name + '.dat'],
    voltage_gain=1.,
    sample_rate=24414,
    n_channels=32,
    dtype='int16',
)

spikedetekt = dict(
    filter_low=500.,  # Low pass frequency (Hz)
    filter_high_factor=0.95 * .5,
    filter_butter_order=3,  # Order of Butterworth filter.

    filter_lfp_low=0,  # LFP filter low-pass frequency
    filter_lfp_high=300,  # LFP filter high-pass frequency

    chunk_size_seconds=1,
    chunk_overlap_seconds=.015,

    n_excerpts=50,
    excerpt_size_seconds=1,
    threshold_strong_std_factor=5.,
    threshold_weak_std_factor=3.,
    detect_spikes='negative',

    connected_component_join_size=1,

    extract_s_before=10,
    extract_s_after=22,

    n_features_per_channel=3,  # Number of features per channel.
    pca_n_waveforms_max=10000,
)

klustakwik2 = dict(
    num_starting_clusters=100,
)
```

ibadr, 2015-08-16:
The line that caused the crash in the comment above is exactly the same line that resulted in a crash in my first comment on #581 (https://github.com/kwikteam/phy/issues/581#issuecomment-131598343).

ibadr, 2015-08-16:
A possibly related error occurs while re-clustering using the API with IPython

```
session=Session(kwik_path,channel_group=1)
session.cluster()
...
INFO     klustakwik: Iteration 115F: 7 clusters, 4 changed, score=-1828819.268405 (decreased by 0.007925)
INFO     klustakwik: Iteration 116F: 7 clusters, 1 changed, score=-1828819.275824 (decreased by 0.007420)
INFO     klustakwik: Iteration 117F: 7 clusters, 0 changed, score=-1828819.275377 (decreased by -0.000447)
INFO     klustakwik: No points changed and last step was full, so trying to split.
INFO     klustakwik: Trying to split clusters
INFO     klustakwik: No points changed, previous step was full and did not split, so finishing.
14:31:55 [I] The automatic clustering process has finished.
14:31:55 [I] Switched to `empty` clustering.
14:31:55 [I] Switched to `main` clustering.
Initializing features and masks: 100.0%.
Features and masks initialized.←[K
Initializing features and masks: 100.0%.
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-75-b1cc71342cde> in <module>()
----> 1 session.cluster()

c:\users\islam\phy\phy\session\session.py in cluster(self, clustering, algorithm, spike_ids, channel_group, **kwargs)
    434         if clustering == 'main':
    435             self.model.copy_clustering('main', 'original')
--> 436         self.change_clustering(clustering)
    437
    438         # Set the new clustering metadata.

c:\users\islam\phy\phy\session\session.py in change_clustering(self, clustering)

    214         self.model.clustering = clustering
    215         info("Switched to `{}` clustering.".format(clustering))
--> 216         self.emit('open')
    217
    218     def on_open(self):

c:\users\islam\phy\phy\utils\event.py in emit(self, event, *args, **kwargs)
    126                 kwargs = {n: v for n, v in kwargs.items()
    127                           if n in argspec.args}
--> 128             res.append(callback(*args, **kwargs))
    129         return res
    130

c:\users\islam\phy\phy\session\session.py in on_open(self)
    217
    218     def on_open(self):
--> 219         self._create_cluster_store()
    220
    221     def on_close(self):

c:\users\islam\phy\phy\session\session.py in _create_cluster_store(self)
    201         # If the cluster store already exists and is consistent
    202         # with the data, it is not recreated.
--> 203         self.store.generate()
    204
    205     def change_channel_group(self, channel_group):

c:\users\islam\phy\phy\io\store.py in generate(self, mode)
    661         debug("Initializing the cluster store for {0:s}.".format(name))
    662         for item in self._items.values():
--> 663             item.store_all(mode)
    664
    665     # Load

c:\users\islam\phy\phy\io\kwik\store_items.py in store_all(self, mode)
    442         if need_generate:
    443             spc = {cluster: self._subset_spikes_cluster(cluster, force=True)
--> 444                    for cluster in clusters_to_generate}
    445
    446             # All spikes to fetch and save in the store.

c:\users\islam\phy\phy\io\kwik\store_items.py in <dictcomp>(.0)
    442         if need_generate:
    443             spc = {cluster: self._subset_spikes_cluster(cluster, force=True)
--> 444                    for cluster in clusters_to_generate}
    445
    446             # All spikes to fetch and save in the store.

c:\users\islam\phy\phy\io\kwik\store_items.py in _subset_spikes_cluster(self, cluster, force)
    390     def _subset_spikes_cluster(self, cluster, force=False):
    391         if force or cluster not in self._spikes_per_cluster:
--> 392             spikes = self._selector.subset_spikes_clusters([cluster])[cluster]
    393             # Persist the new _spikes_per_cluster array on disk.
    394             self._spikes_per_cluster[cluster] = spikes

KeyError: 7
```

Interestingly, even though KlustKwik reported 7 clusters, only 6 clusters got written to the disk:
```
In [76]: session.model.describe()
Kwik file               C:\Users\islam\phytest\TDT\experiment_Block-2.kwik
Recordings              1
List of shanks          0, 1*, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 1
6, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31
Clusterings             main*, empty, original
Channels                1
Spikes                  181276
Clusters                6
Duration                3625s
```

megkirch, 2015-09-17:
Has this issue been resolved? I just upgraded to phy 0.2.2, but I am still getting the exact same error as ibadr above (and the same as I was getting with phy 0.2.1), resulting in:
IndexError: index 82496 is out of bounds for axis 0 with size 66202

I get this same error whether I do session.change_channel_group(1) from ipython, or phy phy cluster-auto --channel-group=1 kwikdata.prm

If I do $ phy cluster-manual amplifier.kwik --channel-group=1, I get:
Traceback (most recent call last):
  File "\miniconda3\lib\runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "\miniconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "\miniconda3\Scripts\phy.exe\__main__.py", line 9, in <module>
  File "\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 434, in main
    out = func(args)
  File "\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 375, in cluster_manual
    use_store=not(args.no_store),
  File "c:\users\megan\miniconda3\lib\site-packages\phy\scripts\phy_script.py", line 248, in _create_session
    session = Session(kwik_path, **kwargs)
  File "\miniconda3\lib\site-packages\phy\session\session.py", line 91, in __init__
    gui_classes=self._gui_classes,
  File "\miniconda3\lib\site-packages\phy\io\base.py", line 387, in __init__
    self.open(path, model=model)
  File "\miniconda3\lib\site-packages\phy\session\session.py", line 136, in open
    return super(Session, self).open(model=model, path=kwik_path)
  File "\miniconda3\lib\site-packages\phy\io\base.py", line 436, in open
    model = self._create_model(path)
  File "\miniconda3\lib\site-packages\phy\session\session.py", line 108, in _create_model
    waveform_filter=self._waveform_filter,
  File "\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 333, in __init__
    clustering=clustering)
  File "\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 729, in open
    self._clustering_changed(self._clustering)
  File "\miniconda3\lib\site-packages\phy\io\kwik\model.py", line 798, in _clustering_changed
    raise ValueError("The clustering {0} is invalid.".format(value))
ValueError: The clustering main is invalid.

Any updates? Thanks in advance for the help!

PR #667 (2015-10-08): [MRG] fix #581 and fix #620
* Fixes #581 
* Fixes #620

Note that when showing a GUI with a new clustering/shank you need to use `session.show_gui()` to create and load a new GUI rather than `gui.show()` which will try and show the old one with disastrous results...
commit 5026798d (2015-10-08): Merge pull request #667 from nippoo/switch_clusterings

[MRG] fix #581 and fix #620
commit 9e2e8c38 (2015-10-08): fix #581 and fix #620
closed 2015-10-08 by rossant