cortex-lab/phy #626: Problem with command line tools

wpettine, 2015-08-20:
**I am on a 64-bit Windows 7 machine. When I go to use phy command with the proper arguments, I get a similar return as if I only entered phy. For example:**

$ phy

too few arguments

usage: phy-script.py [-h] [--version] [--debug] [--hide-traceback]
                     [--profiler] [--line-profiler] [--ipython] [--pdb]

                     {download,traces,describe,spikesort,detect,cluster-auto,cluster-manual}
                     ...

Spike sorting and ephys data analysis for 1000 channels and beyond.

optional arguments:
  -h, --help            show this help message and exit
  --version, -v         print the version of phy
  --debug, -d           activate debug logging mode (default: False)
  --hide-traceback      hide the traceback for cleaner error messages
                        (default: False)
  --profiler, -p        activate the profiler (default: False)
  --line-profiler, -lp  activate the line-profiler -- you need to decorate the
                        functions to profile with `@profile` in the code
                        (default: False)
  --ipython, -i         launch the script in an interactive IPython console
                        (default: False)
  --pdb                 activate the Python debugger (default: False)

subcommand:
  {download,traces,describe,spikesort,detect,cluster-auto,cluster-manual}
    download            download a sample dataset
    traces              show the traces of a raw data file
    describe            describe a `.kwik` file
    spikesort           launch the whole spike sorting pipeline on a `.prm`
                        file
    detect              launch the spike detection algorithm on a `.prm` file
    cluster-auto        launch the automatic clustering algorithm on a `.kwik`
                        file
    cluster-manual      launch the manual clustering GUI on a `.kwik` file

examples:
  phy -v                display the version of phy
  phy download hybrid_120sec.dat -o data/
                        download a sample raw data file in `data/`
  phy describe my_file.kwik
                        display information about a Kwik dataset
  phy spikesort my_params.prm
                        run the whole suite (spike detection and clustering)
  phy detect my_params.prm
                        run spike detection on a parameters file
  phy cluster-auto my_file.kwik
                        run klustakwik on a dataset (after spike detection)
  phy cluster-manual my_file.kwik
                        run the manual clustering GUI

**If I use the download command:**

$ phy download hybrid_10sec.dat

usage: phy-script.py [-h] [--version] [--debug] [--hide-traceback]
                     [--profiler] [--line-profiler] [--ipython] [--pdb]

                     {download,traces,describe,spikesort,detect,cluster-auto,cluster-manual}
                     ...

Spike sorting and ephys data analysis for 1000 channels and beyond.

optional arguments:
  -h, --help            show this help message and exit
  --version, -v         print the version of phy
[... 102 lines cut ...]

launch the manual clustering GUI on a `.kwik` file

positional arguments:
  file                  path to a `.kwik` file

optional arguments:
  -h, --help            show this help message and exit
  --debug, -d           activate debug logging mode
  --hide-traceback      hide the traceback for cleaner error messages
  --profiler, -p        activate the profiler
  --line-profiler, -lp  activate the line-profiler -- you need to decorate the
                        functions to profile with `@profile` in the code
  --ipython, -i         launch the script in an interactive IPython console
  --pdb                 activate the Python debugger
  --clustering CLUSTERING
                        name of the clustering to use
  --cluster-ids CLUSTER_IDS, -c CLUSTER_IDS
                        list of clusters to select initially
  --no-store            do not create the store (faster loading time, slower

**Here's a screenshot of my packages installed in miniconda:**
[image]

**I went through the install process and verified the presence of all dependencies. My console returns "0.2.1" with $ phy -v. I was unable to find a solution on the download site, and haven't seen a similar post elsewhere on the forum. Any suggestions?**

rossant, 2015-08-20:
Hi, it looks like a duplicate of #465
we'll be looking into it very soon!

closed 2015-08-20 by rossant

wpettine, 2015-08-20:
Cool. Please let me know if  there is any way I can help!

rossant, 2015-09-15:
this is fixed in the latest version