cortex-lab/phy #628: Backport file format fixes in KV?

rossant, 2015-08-25:
to make phy-created files compatible with KV? might be simpler for users if things "just worked" rather than having x slightly incompatible versions of the files

nippoo, 2015-08-25:
I guess we should make the waveforms dynamically loaded too, then? happy to do this but it should probably be a KV issue...

rossant, 2015-08-25:
Ahhh you're right for the waveforms. Then it's not that simple, because then you need the raw data, but it's using kwd and not .dat etc etc. Not sure if it's worth it then.

I guess the question is why people want to use KV? Maybe they lack the cluster view?

nippoo, 2015-08-25:
Well the ClusterView, but probably also the SimilarityMatrixView for smaller datasets... perhaps compatibility with .clu files? Or one of a few blocking bugs (kk2 parameters not working... etc)!

nippoo, 2015-08-27:
Closing in favour of https://github.com/klusta-team/klustaviewa/issues/58

closed 2015-08-27 by nippoo