cortex-lab/phy #638: sample parameters file not downloading

nkalipi, 2015-09-10:


rossant, 2015-09-10:
it seems that you have an extra dot at the end of the file: try removing it

nippoo, 2015-09-10:
Looking at the logs, someone has recently requested ```hyrbid_10sec.prm``` which is a typo. Can you check you've got the correct full filename?

nkalipi, 2015-09-10:
i ran the command as given in the website, this is what i am getting:

it says that the file cannot be found at the website specified..
[image]

nippoo, 2015-09-10:
Can you try browse to that URL in your web browser? Have you got some sort of institutional proxy or firewall?

http://phy.cortexlab.net/data/samples/

nkalipi, 2015-09-10:
yep that worked! thanks!! somehow my terminal is not able to point to that website..
Thanks @nippoo

closed 2015-09-10 by rossant