cortex-lab/phy #639: waveform display offset

danieljdenman, 2015-09-10:
i'm seeing an issue with the offset of the extracted waveforms displayed in the Waveform view - they appear to be offset in the amplitude dimension:

[image]

this data was sorted using phy2.1, displayed with 2.2. running on mac with anaconda.

danieljdenman, 2015-09-11:
amended: true for some data sets, not others. working on determining the unifying factor between datasets with this issue.

nippoo, 2015-10-06:
any updates @danieljdenman? maybe the .prb might be different / wrong / there might be a huge DC offset or something.

I'm working on the WaveformView this week so if you haven't figured it out, I'd love to have a copy of a dataset which exhibits this behaviour so I can troubleshoot myself!

danieljdenman, 2015-10-06:
i never quite figured out what was the cause. deleting the store [that was created with 2.1] and remaking it with 2.2 solves the issue for me. since everything is updated now, i haven't seen any issues.

nippoo, 2015-10-06:
alright, closing for now!

closed 2015-10-06 by nippoo