cortex-lab/phy #644: SimilarityMatrix

nippoo, 2015-09-16:
We should, y'know, bring back the SimilarityMatrix at some point.

Or at least something you can pop up and manually recalculate when needed. For low-channel-count probes this is a good feature to have!

rossant, 2015-09-16:
I'm +1 on the idea, but I think it should be significantly different from the old one:

* customizable quality/similarity metrics (this is already implemented). diagonal = quality, out of the diagonal = similarity
* should be *fast* to compute (= instantaneous)
* should be written in HTML: I guess a big `<table>` with background colors for the `<td>`  and `on_click` events or whatever (no point using OpenGL for this!)

rossant, 2019-06-14:
by @nsteinme (https://github.com/kwikteam/phy-contrib/issues/112) :

Similarity view was abandoned when we had recordings with a thousand clusters, but a lot of people liked it. Here's a simple way to bring it back in a format that's usable for everyone: auto-zoom it. When selecting a new cluster, zoom into (e.g.) a 25x25 segment of the matrix centered on that cluster. The rows could be ordered by cluster depth, or by whatever ordering is currently employed in the cluster view, or by similarity with current cluster.

Personally I've never been much bothered by the exclusion of this view, but if people are asking for it then this could be a way to make it sensible.