cortex-lab/phy #645: Show total number and clickable list of clusters in each group

nippoo, 2015-09-16:
Initially let's just have a clickable cluster number to vaguely emulate the functionality of the old ClusterView in a tree. Probably only display 10-20 clusters by default, but users can remove the limit if they have the screen space.

nippoo, 2015-09-16:
cc @nsteinme

rossant, 2015-09-16:
I guess we could even show all clusters: this should be fast in HTML
like a simple `<ul>` list

nsteinme, 2015-09-16:
Agree, just a simple string for each: "123: MUA, 72456" and when you click it, just runs ":c 123". Then maybe ctrl+click a second (third, etc) will additionally select those. Maybe sort by group then cluster number. 

This has been requested by everyone I've talked with who has tried phy! And then we heard today at electrophysiology club that some people (Mush!) will refuse to use phy until this feature exists.

nippoo, 2015-09-16:
Almost-duplicate of #613

rossant, 2015-09-23:
Actually a `<table>` might be better than a bullet list...

closed 2016-03-24 by rossant