cortex-lab/phy #649: spikes are not being properly assigned to recordings > 0

neuromusic, 2015-09-25:
Nearly all of my experiments are multi-recording files that I need to sort units across recordings.

I can run the auto clustering and it seems to detect & cluster spikes on all recordings. However...

1. Every spike has been assigned to recording 0 (that is, `/channel_groups/0/spikes/recording` is full of zeros)
2. Every spike is sampled relative to recording 0 (that is, `/channel_groups/0/spikes/time_samples` is monotonically increasing, with values up to 26067136 when the length of recording 0 is only 801364 samples)

Note: this is using the latest commit https://github.com/kwikteam/phy/commit/487b6b1a4bbb73848531729aedd488f7bb35c8f9

neuromusic, 2015-09-25:
is this the problem? 

https://github.com/kwikteam/phy/blob/7d01c66f4d8ada8024ed2def92bef43ce8b3b544/phy/session/session.py#L337

nippoo, 2015-09-25:
Yeah, it's basically a duplicate of https://github.com/kwikteam/phy/issues/634. @rossant is working on this now - we're changing the whole I/O part to allow a lot more user-flexibility. We've realised that maintaining a rigid format isn't really what users want. The "recordings" array is silly - you've basically got an array of len(n_spikes) which generally contains at most 8 bits of total entropy, ha!

nippoo, 2015-09-25:
So at the moment your best bet is to concatenate the files in the OS or somehow else - the next release (0.3.0) will support all this and more. Sorry!

nippoo, 2015-09-25:
https://github.com/kwikteam/phy/pull/633 is all we've 'done' about it so far.

closed 2016-03-24 by rossant