cortex-lab/phy #653: running phy on windows 10

justjul, 2015-09-28:
I get the following error when running phy on my windows 10 machine (64 bits, Intel Xeon):

D:\DATA\phy_test>phy spikesort hybrid_10sec.prm
Traceback (most recent call last):
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\Scripts\phy-script.py", line 9, in <module>
    load_entry_point('phy==0.2.2', 'console_scripts', 'phy')()
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\lib\site-packages\phy\scripts\phy_script.py", line 434, in main
    out = func(args)
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\lib\site-packages\phy\scripts\phy_script.py", line 351, in spikesort
    kwik_path=kwik_path,
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\lib\site-packages\phy\io\kwik\creator.py", line 385, in create_kwik
    prm = _read_python(prm_file) if prm_file else {}
  File "C:\Program Files\KlustaViewa\WinPython-64bit-2.7.6.4\python-2.7.6.amd64\lib\site-packages\phy\utils\_misc.py", line 129, in _read_python
    assert op.exists(path)
AssertionError

Any idea where that could come from?
phy seems to be well installed (I can get the help summary if I type in phy)
Note also that the download function of phy didn't work either (I had to download the test dataset manually)

nippoo, 2015-10-06:
Hi - can you make sure there's a file called hybrid_10sec.prm and hybrid_10sec.dat which exist in the same directory? This looks like it just can't find the file.

justjul, 2015-10-06:
thanks
windows had indeed added a .txt extension when I edited the prm file. Removing this extension solved the problem.

nippoo, 2015-10-06:
:)

closed 2015-10-06 by nippoo